PX4/PX4-Autopilot #22121: [Bug] the aircraft in QGC and gazebo are not moving after the preflight check was OK

zyy37, 2023-09-21:
### Describe the bug

[image]
[image]

Using the software versions of px4(`px4_sitl`), `gazebo garden` and QGC for simulation, the preflight check was OK, but the aircraft in QGC and gazebo are not moving.
From the above figure, it can be seen that after 5 minutes and 30 seconds, the aircraft in QGC did not move, and so did the aircraft in Gazebo.

### To Reproduce

1. `make px4_sitl gz_x500`
2. connect to the QGC by UDP.

### Expected behavior

the aircraft in QGC and gazebo should follow the route and move after the preflight check was OK.

### Flight Log

n/a

### Software Version

ver all
HW arch: PX4_SITL
PX4 git-hash: d2819ca7542c7fc1c3b122371d9dd35cb7005270
PX4 version: 1.14.0 80 (17694848)
PX4 git-branch: main
OS: Linux
OS version: Release 5.15.0 (84869375)
Build datetime: Sep 19 2023 12:14:40
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 11.4.0
PX4GUID: 1006554d49534954414c44494e4f30303030
UNKNOWN MCU

ubuntu@ubuntu:~$ gz sim --version
Gazebo Sim, version 7.0.0
Copyright (C) 2018 Open Source Robotics Foundation.
Released under the Apache 2.0 License.

### Flight controller

px4_sitl

### Vehicle type

None

zyy37, 2023-10-01:
Any update? Please. Is it a bug?
here is the log if you want to know. https://logs.px4.io/plot_app?log=191b9853-ee13-4475-b7d5-b6a134eaefc3

bresch, 2023-10-02:
Is it a custom model? Looking at the log, the drone went full thrust but didn't move.

zyy37, 2023-10-03:
Not a custom model.
I'm new to px4 and just want to run an default example. The development environment I am using is Ubuntu 22.04 on Raspberry Pi 4b, and the QGroundControl environment is Windows 10. I followed the official steps which are [Ubuntu Development Environment](https://docs.px4.io/main/en/dev_setup/dev_env_linux_ubuntu.html) and [Building PX4 Software](https://docs.px4.io/main/en/dev_setup/building_px4.html) to reproduce it. However, when running this example `make px4_sitl gz_x500`, it shows the above result that the plane does not move, and I don't know where the problem is.

zyy37, 2023-10-08:
[image]
Fig 1. UDP settings of referring website.
[image]
Fig 2. The moving aircraft of referring website.

[image]
Fig 3. I cannot reproduce it, and the aircraft is not moving. `make px4_sitl gz_x500`

Both are done by running px4_sitl and gazebo, as well as QGroundControl linked by UDP, I imitated the settings and steps in Figure 1 and Figure 2 (referring to the [website](https://www.youtube.com/watch?v=iVU8ZNoMn_U)) step by step, but the aircraft still couldn't move. The blogger's aircraft can move, but mine cannot, but he uses the gazebo classic and I use the gazebo garden. I cannot reproduce it. It's very strange. Is this a bug?

zyy37, 2023-10-29:
Update:
When the development machine is replaced with a Linux amd64 virtual machine, the aircraft in the software simulation (`make px4_sitl gz_x500`) can be moved. The previous development machine for the aircraft that did not move was the Raspberry Pi 4b 8GB arm64. You may support this scenario. thanks

[image]

By the way, the error logs for aircraft not moving on the Raspberry Pi 4b 8GB arm64 are these,
[image]

AkashJohnSubash, 2024-08-15:
Suspect its a difference in the binary releases of one of the packages for arm64. (Same setup worked fine for amd64)
I had a similar issue using Gazebo Harmonic running on Ubuntu Jammy for apple Silicon (m1Pro) with [CrazySim_SITL](https://github.com/gtfactslab/CrazySim). 
I Would be interested in tips on debugging to find the offending package.