PX4/PX4-Autopilot #22122: [Bug] Critical: Arming denied: high throttle

zyy37, 2023-09-21:
### Describe the bug

[image]
[image]
In hardware simulation, virtual joystick are used. Even if there are no issues with the pre flight inspection, the aircraft cannot arm. Critical: Arming denied: high throttle.

Can your team provide feasible solutions or tips while printing preflight check errors? For example, for CBRK_SUPPLY_CHK, the tip is checking if there is battery power or just disable it. Otherwise, users spend a lot of time searching for the pre flight errors and solving, and are stuck here all the time. Because no one understands internal logic and solutions better than your team. Thanks.

### To Reproduce

1. Install QGC daily on Windows 10 x64.
2. Plug pixhawk 2.4.8 into the computer port via USB.
3. firmware setup.

### Expected behavior

The aircraft preflight check turning green means it should be able to arm and fly.

### Flight Log

n/a

### Software Version

nsh> ver all
HW arch: PX4_FMU_V3
HW type: V2
HW version: 0x90008
HW revision: 0x000
PX4 git-hash: d1266c856fbd759cbc6cf583c5221aab49962b30
PX4 version: 1.14.0 80 (17694848)
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: c23b72dffeb0de0d1a836ab561eb9169c4a9e58e
Build datetime: Sep 14 2023 15:11:12
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000100000000303532353438511400270039
MCU: STM32F42x, rev. 3

### Flight controller

pixhawk 2.4.8

### Vehicle type

None

danielhonies, 2023-10-04:
You need to pull the virtual joystick down to arm. Alternatively you can connect a gamepad. The virtual joysticks are sending PWM of 1500 in default for throttle so it won't arm.

julianoes, 2023-10-05:
Or have you tried arming in a mode such as mission or position control? Does that work?

zyy37, 2023-10-08:
> You need to pull the virtual joystick down to arm. Alternatively you can connect a gamepad. The virtual joysticks are sending PWM of 1500 in default for throttle so it won't arm.

Thank you for your reply. At that time, I plugged pixhawk into QGroundControl on Windows 10, and the default mode should be real flight mode, not HITL mode, but this should does not matter. But when I click down on the virtual joystick on the left or right, the throttle is still very high, indicating that the background data has not been effective.

Later, I followed the steps of [Hardware in the Loop Simulation (HITL)](https://docs.px4.io/main/en/simulation/hitl.html) in Figure 1 and enabled HITL, but the aircraft still cannot move. I don't know if /dev/ttyACM0 needs to be specified here, but I am using the gazebo garden on Ubuntu 22.04, which is not in the specified simulators (Gazebo Classic and jMAVSim).

The above are two questions.

Additionally, I think the virtual joystick scenario is also very important. Because people don't want to carry too many things for the convenience of mobility.

[image]
Fig 1. [Hardware in the Loop Simulation (HITL)](https://docs.px4.io/main/en/simulation/hitl.html)

zyy37, 2023-10-08:
> Or have you tried arming in a mode such as mission or position control? Does that work?

Thank you for your reply. At that time, I plugged pixhawk into QGroundControl on Windows 10, and the default mode should be real flight mode, not HITL mode. Because the 'AltitudeRelative' displayed on the QGroundControl panel and the fast moving pixhawk 2.4.8 board seem to respond roughly correctly, which means it is in real flight mode. In that scene, other modes such as position mode or mission mode may require real environmental data as forbidden by reporting errors: no valid local / global position estimate.

However, the manual mode still reports errors which are MAV_CMD_COMPONENT_ARM_DISARM and Arming denied: high throttle after pulling down the virtual joystick, indicate that the throttle data in the background has not been reduced. See [comment](https://github.com/PX4/PX4-Autopilot/issues/22122#issuecomment-1752163140) for details.

Gnaneswar484, 2023-10-19:
Trying to fly the drone to desired position using code in qgc and jmavsim.

I want to move drone to particular position using the code below mentioned, I build the code successfully into px4-firmware and when running the code using jmavsim it was successful, but the drone was not moving. Can anyone of you check is there any problem with the code as I am new to the px4-development, I am not able to debug it.

#include <px4_platform_common/px4_config.h>
#include <px4_platform_common/tasks.h>
#include <uORB/uORB.h>
#include <uORB/topics/vehicle_local_position_setpoint.h>
#include <uORB/topics/vehicle_local_position.h>

extern "C" __EXPORT int my_autonomous_flight_module_main(int argc, char *argv[]);

static volatile sig_atomic_t _should_exit = false;
static int _task = -1;

// Target position (latitude, longitude, altitude)
static constexpr double target_pos[] = {47.398036, 8.545484, 5.0}; // Replace with your target position

int my_autonomous_flight_module_thread(int argc, char *argv[]) {
PX4_INFO("My Autonomous Flight Module is running!");

orb_advert_t pos_sp_adv = orb_advertise(ORB_ID(vehicle_local_position_setpoint), nullptr);

// Main loop
while (!_should_exit) {
    // Create a new setpoint
    struct vehicle_local_position_setpoint_s sp{};
    sp.x = target_pos[0];
    sp.y = target_pos[1];
    sp.z = target_pos[2];

    // Publish the setpoint
    orb_publish(ORB_ID(vehicle_local_position_setpoint), pos_sp_adv, &sp);

    px4_usleep(1000000); // Sleep for 1 second
}

PX4_INFO("My Autonomous Flight Module exiting.");
return 0;
}

extern "C" int my_autonomous_flight_module_main(int argc, char *argv[]) {
// Start the module's main thread
_task = px4_task_spawn_cmd(
"my_autonomous_flight_module",
SCHED_DEFAULT,
SCHED_PRIORITY_DEFAULT + 40,
2000,
my_autonomous_flight_module_thread,
(char *const *)nullptr);

if (_task < 0) {
    PX4_ERR("Task start failed");
    return -errno;
}

return 0;
}

Gnaneswar484, 2023-10-19:
Are there any other methods to fly drone from one point to another using build code

MaEtUgR, 2023-12-12:
@zyy37 Was your problem resolved by the answers?

The check which fails with `Arming denied: high throttle` is a feature to prevent someone accidentally arming in manual mode while giving high throttle. E.g. when you do that on a multicopter it shoots up into the air without any further input and you have to immediately steer it. There is no circuit breaker for it but it should be no issue to just hold the throttle down while arming e.g. by:
- Arming using the stick gesture where you move the virtual joystick's left stick into the bottom right corner. This is the gesture which when hold for 1 second arms the vehicle and at the same time your throttle is low.
- If you want to arm using the arm command from the UI in manual mode with the virtual joysticks enabled you can park their position with a low throttle by moving the left virtual joystick all the way down and while holding it switch window using [Alt] + [Tab]. The stick's position stays there until you move it for the next time.

For reference here's the implementation of the check:
https://github.com/PX4/PX4-Autopilot/blob/91ab09ebd5d86390912ebc51bb506bc6ad061394/src/modules/commander/Commander.cpp#L575-L584

If you're still experiencing problems please provide further details:
Are you trying to fly a HITL quadrotor? What mode do you plan to fly in? Why can't you switch to that mode before arming? Why can't you arm using the gesture?

zyy37, 2023-12-13:
Your two solutions are okay. It's worked. You considered it very accurately because it seems impossible for people to operate two clicks at the same time on a computer, like clicking armed and virtual joystick's low throttle. Thank you.