PX4/PX4-Autopilot #22124: [Bug] JointPositionController not working during px4 SITL

Alessioo1, 2023-09-22:
### Describe the bug

JointPositionController is not working during px4 SITL , the command is recieved by gazebo but the joint is not moving.
The same model.sdf in a gazebo world with the SITL NOT RUNNING is working correctly

### To Reproduce

1. Start ros_gz_bridge-parameter_bridge
2. Start MicroXRCEAgent udp4 -p 8888
3. Start SITL : PX4_SYS_AUTOSTART=40522 PX4_GZ_MODEL=qav_2 PX4_GZ_WORLD=blocks PX4_GZ_MODEL_POSE="0 0 0.34 0 0 0" ./build/px4_sitl_default/bin/px4
4. Start ros2 control node
5. trigger the node to send command on the specific topic

### Expected behavior

Joint should move

### Screenshot / Media

[videos.zip](https://github.com/PX4/PX4-Autopilot/files/12700974/videos.zip)

### Flight Log

https://review.px4.io/plot_app?log=8d417341-f3e5-4832-8d97-52a298559894

### Software Version

ver all
HW arch: PX4_SITL
PX4 git-hash: 4fce159a2ab490410e730d492d4135f1beca2db4
PX4 version: 1.14.0 0 (17694720)
PX4 git-branch: main
OS: Linux
OS version: Release 5.15.90 (84892415)
Build datetime: Sep 22 2023 09:01:06
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 11.4.0
PX4GUID: 1006554d49534954414c44494e4f30303030
UNKNOWN MCU
pxh> INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_global_position data writer, topic id: 219
INFO  [commander] [32mReady for takeoff![0m

### Flight controller

px4

### Vehicle type

Multicopter

Jaeyoung-Lim, 2023-09-22:
@Alessioo1 What is quav_2?

Alessioo1, 2023-09-22:
It's a custom model 
[qav_2.zip](https://github.com/PX4/PX4-Autopilot/files/12701167/qav_2.zip)
the airframe is 
[airframe custom.zip](https://github.com/PX4/PX4-Autopilot/files/12701186/airframe.custom.zip)

Jaeyoung-Lim, 2023-09-22:
I won't be able to check the sdf file myself, but you can try to findout whether the joint names are named correctly to what px4 is publishing

Alessioo1, 2023-09-22:
I'm able to see the data published in gazebo with an echo of the topic, isn't enough to check that px4 is publishing correctly?

Alessioo1, 2023-09-22:
Yes it is. I use the same model.sdf when I spawn the robot without the sitl and it's working, doesn't it mean that the plugins are configured correctly?

Jaeyoung-Lim, 2023-09-22:
But if the joint control topics are publishing to the right joints, why aren't the joints moving?

Alessioo1, 2023-09-22:
I don't know, that's why I opened the issue. I have also tried to change the joint names because I had named them "servo_1" and "servo_2" and I thought that that could cause some type of conflict/error inside sitl. Nothing has changed changing names.

Jaeyoung-Lim, 2023-09-22:
>  I have also tried to change the joint names because I had named them "servo_1" and "servo_2"

What did you change it to?

Alessioo1, 2023-09-23:
“serrrvo_1” and “serrrvo_2”

Jaeyoung-Lim, 2023-09-23:
@Alessioo1 Exactly, is px4 publishing a joint control topic to that joint?

Alessioo1, 2023-09-23:
From my ROS2 node I'm publishing messages to the input topic of the jointPositionController plugin, and ros_gz_bridge is reporting the topic in gazebo and the plugin should receive it, am I missing something? Is px4 involved in these steps? The names of the topics are correct I've changed them accordingly in each file: model.sdf , ros2 publisher/subscriber and  ros_gz_bridge .
As before everything works outside sitl but not with sitl running.

Alessioo1, 2023-10-05:
Here it is the modified model of qav_2 with relative path for meshes files
[qav_2.zip](https://github.com/PX4/PX4-Autopilot/files/12815124/qav_2.zip)

Alessioo1, 2023-10-07:
Solved, for some reason inside the simulation the model qav_2 was named qav_2_0 changing the topic of jointpositioncontroller

closed 2023-10-07 by Alessioo1