PX4/PX4-Autopilot #22128: [Bug] there is no sensor_combined data in some time periods

huang-shijie, 2023-09-24:
### Describe the bug

there is no sensor_combined data in some time periods

### To Reproduce

I am the flight controller of cuav v5. I downloaded the 1.14 firmware, ran ros2 bag record, recorded the topics in the picture, and then used the rqt_bag command to display them. I found that the data released by these topics related to imu data are There is no data in some time periods, which makes me directly subscribe to sensor combined for the case of vio drift and crash. I don’t know the reason for this.
[image]

### Expected behavior

1

### Flight Log

1

### Software Version

1

### Flight controller

1

### Vehicle type

None

suezere, 2023-12-20:
Hi I’m trying to fly by building V1.14 (beta) on Cuav v5+. But here comes the problem. After building the beta version, the FC status color changes to red. I would like to ask if you have encountered this situation and how to solve it. Thanks.

beniaminopozzan, 2026-09-29:
I've observed the same behaviour when using serial protocol