PX4/PX4-Autopilot #22130: [Bug] v1.14.0-rc1 version rover airframe bug

eogks8921, 2023-09-25:
### Describe the bug

Hello. I am conducting research related to rovers using Pixhawk6X. I was trying to carry out my research with firmware version 1.14.0-rc1, but I encountered an issue where there was no PWM change after selecting the airframe and attempting to arm. While inspecting the source code, I found a bug.

In the following code:

PX4-PX4-Autopilot/boards/px4/fmu-v6x/default.px4board

I noticed that CONFIG_MODULES_ROVER_POS_CONTROL=y was missing.

### To Reproduce

Error in the 1.14.0-rc1 firmware version itself.

### Expected behavior

Even after selecting the rover airframe, arming, and changing the throttle, there is no PWM output change.

### Screenshot / Media

[image]

### Flight Log

None

### Software Version

1.14.0-rc1

### Flight controller

Pixhawk6X

### Vehicle type

Rover

giangthewalkingman, 2023-11-04:
same, my rover model use  skid-steering, I use main 1 and main 2 to connect to those ESCs. And, when I run Actuator testing at Actuator setup menu of QGcontrol, the motors act as normal. However, when I use RC control or try controlling at the main menu (using those sticks in the display), the PWM out keeps at 1500 though I move those sticks. When the firmware at 1.13.3, my rover acts normally. I think it is a bug.

AlbertoJimenezUC3M, 2024-02-28:
Hello, has anyone been able to solve it? I can´t use a rover with 1.14 :(

eogks8921, 2024-02-28:
@AlbertoJimenezUC3M hello, to solve this problem. you need to write CONFIG_MODULES_ROVER_POS_CONTROL=y in default.px4board and build again. If you face some kind of memory error, delete no use modules.

AlbertoJimenezUC3M, 2024-03-07:
@eogks8921 Thanks for your help, it worked fine!

maxpolzin, 2024-05-16:
@AlbertoJimenezUC3M: Thx for raising the issue, I ran into the same issue. 
@eogks8921 Thx for your inputs, that indeed resolves the problem.

chfriedrich98, 2024-09-13:
Cleaning up rover related issues:
The rover support has since been completely reworked and now uses its own [build](https://docs.px4.io/main/en/frames_rover/#flashing-the-rover-build). 
Rover modules are no longer included in the default build due to flash constrains.

closed 2024-09-13 by chfriedrich98