PX4/PX4-Autopilot #22139: [Bug Report] Out-Of-Boundary Bug Found in /platforms/nuttx/NuttX/apps/examples/gps/gps_main.c

Pwn9uin, 2023-09-26:
### Describe the bug

### Summary

We identified a out of boundary bug in /platforms/nuttx/NuttX/apps/examples/gps/gps_main.c:69 due to the absence of the limited count (i.e., index variable of line buffer).

we reported this vulnerability to PX4/PX4-Autopilot and PX4/NuttX-apps

### Detailed Root Cause

```cpp
do
  {
    read(fd, &ch, 1);
    if (ch != '\r' && ch != '\n')
      {
        line[cnt++] = ch;
      }
  }
while (ch != '\r' && ch != '\n');
```

In /platforms/nuttx/NuttX/apps/examples/gps/gps_main.c:64, there is the code which puts the input data from *read* function to *line* variable. Even though the *line* variable is defined its size as *MINMEA_MAX_LENGTH* which is 256, the cnt doesn’t have any restriction or limitation. so this code will receive the data until ‘\r’ and ‘\n’. If the data does not include '\r' or '\n’ until its size exceeds 80, then out of boundary bug can be triggered.

### To Reproduce

### POC reproduce

**Environment setting**

we checked this vulnerability actually impact on real device.

In order to apply the NuttX-app/examples/gps code to PX4 pixhawk 6c device, we should set the environment as following.

- `git clone [https://github.com/kosma/minmea.git](https://github.com/kosma/minmea.git)` in `platforms/nuttx/NuttX/apps/gpsutils/minmea/minmea/`
- `git reset --hard db46128e73cee26d6a6eb0482dcba544ee1ea9f5` in `platforms/nuttx/NuttX/apps/gpsutils/minmea/minmea/minmea`
- Disable codes upon `$(MINMEA_UNPACKDIR):` rule in `platforms/nuttx/NuttX/apps/gpsutils/minmea/Makefile`
- Set the default of `config EXAMPLES_GPS` as 'y' in `platforms/nuttx/NuttX/apps/examples/gps/Kconfig`
- Add a setting `CONFIG_EXAMPLES_GPS=y` on default.px4board (e.g.`boards/px4/fmu-v6c/default.px4board`)
- Replace `#include "gpsutils/minmea.h"` to `#include "minmea/minmea.h"` (platforms/nuttx/NuttX/apps/examples/gps/gps_main.c:32)
- Replace `MINMEA_MAX_LENGTH` to `MINMEA_MAX_SENTENCE_LENGTH` (platforms/nuttx/NuttX/apps/examples/gps/gps_main.c:47)
- Set `/dev/ttyS1` to `/dev/ttyS0` (Because the correct gps serial is /dev/ttyS0 for PX4_fmu-v6c board)

And just connect the gps module to the port at /dev/ttyS0 then the module get the gps data and the vulnerable code will be triggered.

### Expected behavior

When people use Nuttx-app/examples/gps, then the Buffer overflow can occur so that the drone can be downed.

### Screenshot / Media

### POC reproduction video

This video shows that the real pixhawk device can be affected by the vulnerable code which makes it stop.

https://github.com/PX4/PX4-Autopilot/assets/77731571/a40abd7a-9902-4b48-8bf6-e9b39110d964

### Flight Log

we update this soon.

### Software Version

1.14.0

### Flight controller

pixhawk 6c

### Vehicle type

None

mrpollo, 2023-09-26:
Hey @davids5 can you please verify?

julianoes, 2023-10-17:
The NuttX gps app is not used in PX4.

This issue would have to be filed against https://github.com/apache/nuttx-apps.

closed 2023-10-17 by julianoes