PX4/PX4-Autopilot #22144: Yaw rotation in offboard with Vz > 10 m/s

awaisk65, 2023-09-27:
### Describe the bug

Hello
I am flying a drone in OFFBoard mode. I send velocity and yaw commands to /mavros/setpoint_raw/local topic through MAVROS. When I command a yaw setpoint with Vx and Vy (Vz = 0) there is no issues upto 20 m/s. After inducing Vz (dive is initiated), there is a significant anti-clock yaw rotation. The rotation is not significant at lesser resultant velocities. Here is an example log file:

05 m/s speed ==> https://review.px4.io/plot_app?log=84c55a0b-3ea7-46c7-82ca-8c0cad320b23
10 m/s speed ==> https://review.px4.io/plot_app?log=ea5202e4-2651-4172-b767-cfb47c5dd1ef
15 m/s speed ==> https://review.px4.io/plot_app?log=e7b53f4a-ae12-40c1-9293-435c01559f52
Please look at around 6:26 for 1, 3:08 for and 4:55 for 3, which is the moment when Vz is induced.

In 3, you will exactly see that the yaw_setpoint remains the same but the yaw_estimated significantly decreases which in result induces a deviation in yaw.

If anybody could provide insight on what issues are apparent from the logs or on a solution using a mixer file, I will appreciate it.

Quadrotor Details:
X - configuration,
Using external compass,
EKF2 Estimator,
px4 : v1.11.2

Onboard Computer Details:
ROS melodic,
RPi 4B 8Gb

Thanks!

### To Reproduce

1. Changed MPC parameters as follow:
MPC_VEL_MANUAL = 20
MPC_XY_CRUISE = 26
MPC_XY_VEL_MAX = 26
MPC_Z_VEL_MAX_DN = 26
2. Drone took off using offboard mode with height setpoint through /mavros/setpoint_position/local topic inducing only z = desired height.
3. Drone was cruised to a target position using /mavros/setpoint_raw/local topic with Vx, Vy and yaw ignoring other parameters of the topic.
4. Drone was dived using /mavros/setpoint_raw/local topic with Vx, Vy, Vz and yaw. At this point after inducing Vz, drone deviated from the yaw setpoint in a anti-clock direction.

### Expected behavior

The drone should not deviate from yaw setpoint while following the velocity setpoints

### Screenshot / Media

[image]

### Flight Log

Links of the trials with 3 different speeds:
1. 05 m/s speed ==> https://review.px4.io/plot_app?log=84c55a0b-3ea7-46c7-82ca-8c0cad320b23
2. 10 m/s speed ==> https://review.px4.io/plot_app?log=ea5202e4-2651-4172-b767-cfb47c5dd1ef
3. 15 m/s speed ==> https://review.px4.io/plot_app?log=e7b53f4a-ae12-40c1-9293-435c01559f52

### Software Version

v1.11.2

### Flight controller

pixhawk 4

### Vehicle type

Multicopter

awaisk65, 2023-09-27:
I have tried to resolve the issue by changing from v1.11.2 to v1.11.3 px4 version, 02 out 11 trials showed the exact same issue of yaw estimated being deviated from yaw setpoint but the deviation was significantly lesser than the v1.11.2. Below is the link of v1.11.3 trial flight log with yaw deviation:

https://logs.px4.io/plot_app?log=3ca63f1b-a57b-47e5-bf19-0ea92240cac6

ArsalanKhan22, 2023-10-03:
I also face the same issue. Is there any solution?

bresch, 2023-11-13:
While going down really quickly, the motors RPMs are really low and this is why the drone starts to loose yaw control. To let the control allocator increase the collective thrust to keep some yaw authority, you can enable [airmode](https://docs.px4.io/main/en/config_mc/pid_tuning_guide_multicopter.html#airmode-mixer-saturation) (set it to roll/pitch/yaw).

awaisk65, 2023-12-05:
> While going down really quickly, the motors RPMs are really low and this is why the drone starts to loose yaw control. To let the control allocator increase the collective thrust to keep some yaw authority, you can enable [airmode](https://docs.px4.io/main/en/config_mc/pid_tuning_guide_multicopter.html#airmode-mixer-saturation) (set it to roll/pitch/yaw).
> 
> However, make sure to check the actual vertical terminal velocity of the drone as it might have too much drag to reach those 10m/s with airmode throttling up to maintain a decent stability.

I have solved the issue by limiting the downward angle to 30° in the offboard code (i.e when the Vz component is included the drone will maintain 30° angle with respect to ground).

closed 2023-12-05 by awaisk65