PX4/PX4-Autopilot #22148: [Bug]

LukyKozlojc, 2023-09-27:
### Describe the bug

I have just ran sudo apt upgrade and then I tried to build make px4_sitl gz_x500 and it suddenly stopped working. I keep getting this Error message: 
INFO  [gz_bridge] world: default, model name: x500_0, simulation model: x500
ERROR [gz_bridge] timed out waiting for clock message
ERROR [gz_bridge] Task start failed (-1)
ERROR [init] gz_bridge failed to start
ERROR [px4] Startup script returned with return value: 256

I tried updating my px4-SITL, clean build and I am still lost.

Does anyone know, what could be a problem here?

### To Reproduce

1. sudo apt upgrade
2. make px4_sitl gz_x500

### Expected behavior

opened simulation (which it does) with working command line to write "commander takeoff".

### Flight Log

I dont have flight log

### Software Version

latest main branch - 1.13.3

### Flight controller

I am using only simulation yet

### Vehicle type

None

dieter-aerit, 2023-09-28:
I have the same thing happening. Docker file to build a simulator environment, suddenly stopped working.
I didn't pin the version of the PX4 software it checks out and builds, I think that's the only variable in my docker file.

Do we have any idea when this broke? Must be a change in the last few weeks ...
I tried to revert back to v1.13.3, but running into other issues right now...

Jaeyoung-Lim, 2023-09-28:
@LukyKozlojc @dieter-aerit If you ran `sudo apt upgrade`, probably one of the dependency versions are out of sync.

LukyKozlojc, 2023-09-28:
Is there a way to find out which one?

Jaeyoung-Lim, 2023-09-28:
@LukyKozlojc You can try to increase the verbosity level of the gz simulator

KevinJeulink1, 2023-09-28:
The solution that worked for me: increase the time out on line 285, to e.g. 5seconds:

https://github.com/PX4/PX4-Autopilot/blob/0df513415609e2eb3bcdaa77f9aeb9665e0c38c3/src/modules/simulation/gz_bridge/GZBridge.cpp#L278-L290

Jaeyoung-Lim, 2023-09-28:
@KevinJeulink1 Although I think your solution is not related to the issue reported in this thread, it would be interesting to understand why that is influencing your setup.

Could you post an issue with the detailed log?

KevinJeulink1, 2023-09-29:
@Jaeyoung-Lim The issue reported by @LukyKozlojc , is actually the same as my issue, I freshly installed PX4 and try to run the gz_x500 simulation. Then the following error in my console appears:

make px4_sitl gz_x500
[0/1] cd /home/kevin/Work/kolibri_px4_...x4_main/build/px4_sitl_default/bin/px4

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=4001
INFO  [param] selected parameter default file parameters.bson
INFO  [param] selected parameter backup file parameters_backup.bson
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 4001
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG0_PRIO: curr: -1 -> new: 50
  CAL_MAG1_ID: curr: 0 -> new: 197644
  CAL_MAG1_PRIO: curr: -1 -> new: 50
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
INFO  [dataman] data manager file './dataman' size is 7866664 bytes
INFO  [init] starting gazebo with world: /home/kevin/Work/kolibri_px4_main/Tools/simulation/gz/worlds/default.sdf
WARN  [init] PX4_GZ_MODEL_NAME or PX4_GZ_MODEL not set using PX4_SIM_MODEL.
INFO  [gz_bridge] world: default, model name: x500_0, simulation model: x500
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
ERROR [gz_bridge] timed out waiting for clock message
ERROR [gz_bridge] Task start failed (-1)
ERROR [init] gz_bridge failed to start
ERROR [px4] Startup script returned with return value: 256
^Cninja: build stopped: interrupted by user.
make: *** [Makefile:232: px4_sitl] Interrupt

After changinig the sleep count, I don't have this issue anymore.

KevinJeulink1, 2023-09-29:
@Jaeyoung-Lim , I do have the same issue as @LukyKozlojc , see my console output below. After increasing the sleep count limit, this issue is solved.

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=4001
INFO  [param] selected parameter default file parameters.bson
INFO  [param] selected parameter backup file parameters_backup.bson
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 4001
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG0_PRIO: curr: -1 -> new: 50
  CAL_MAG1_ID: curr: 0 -> new: 197644
  CAL_MAG1_PRIO: curr: -1 -> new: 50
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
INFO  [dataman] data manager file './dataman' size is 7866664 bytes
INFO  [init] starting gazebo with world: .../Tools/simulation/gz/worlds/default.sdf
WARN  [init] PX4_GZ_MODEL_NAME or PX4_GZ_MODEL not set using PX4_SIM_MODEL.
INFO  [gz_bridge] world: default, model name: x500_0, simulation model: x500
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
ERROR [gz_bridge] timed out waiting for clock message
ERROR [gz_bridge] Task start failed (-1)
ERROR [init] gz_bridge failed to start
ERROR [px4] Startup script returned with return value: 256
^Cninja: build stopped: interrupted by user.
make: *** [Makefile:232: px4_sitl] Interrupt

dieter-aerit, 2023-09-29:
+1 on increasing the sleep fixes the issue.

beniaminopozzan, 2023-10-07:
@LukyKozlojc 
you cannot be using v1.13.3
which branch or tag are you actually using?

jmackay2, 2023-10-11:
Looks like the suggested solution of increasing the sleep has been commit. I no longer see the issue after #22194 has been merged.

closed 2023-10-11 by Jaeyoung-Lim

Rainoxer, 2025-05-01:
> The solution that worked for me: increase the sleep_count_limit on line 285, to e.g. 5seconds:
> 
> [PX4-Autopilot/src/modules/simulation/gz_bridge/GZBridge.cpp](https://github.com/PX4/PX4-Autopilot/blob/0df513415609e2eb3bcdaa77f9aeb9665e0c38c3/src/modules/simulation/gz_bridge/GZBridge.cpp#L278-L290)
> 
> Lines 278 to 290 in [0df5134](/PX4/PX4-Autopilot/commit/0df513415609e2eb3bcdaa77f9aeb9665e0c38c3)
> 
>  		_object.store(instance); 
>  		_task_id = task_id_is_work_queue; 
>   
>  		if (instance->init() == PX4_OK) { 
>   
>  #if defined(ENABLE_LOCKSTEP_SCHEDULER) 
>  			// lockstep scheduler wait for initial clock set before returning 
>  			int sleep_count_limit = 1000; 
>   
>  			while ((instance->world_time_us() == 0) && sleep_count_limit > 0) { 
>  				// wait for first clock message 
>  				system_usleep(1000); 
>  				sleep_count_limit--;

This problem is still present in 1.15 when trying to load "complex" worlds, other than the default one.
But the file changed a bit :

`#if defined(ENABLE_LOCKSTEP_SCHEDULER)
			// lockstep scheduler wait for initial clock set before returning
			int sleep_count_limit = 10000;

			while ((instance->world_time_us() == 0) && sleep_count_limit > 0) {
				// wait for first clock message
				system_usleep(10000); //Was previously at 1000
				sleep_count_limit--;
			}

			if (instance->world_time_us() == 0) {
				PX4_ERR("timed out waiting for clock message");
				instance->request_stop();
				instance->ScheduleNow();

			} else {
				return PX4_OK;
			}`

I had to replace `system_usleep(1000);` with `system_usleep(10000);` (line 338) and it's working fine now.

Meowchine1, 2025-07-23:
Faced a problem with Gazebo and px4 interaction: gz_bridge failed to start and spawn model. Can anyone help? 
```
 MicroXRCEAgent udp4 -p 8888
 gz sim -v 4 aruco_field.sdf
 export PX4_GZ_MODEL_NAME=x500_vision_0 && 
 export PX4_GZ_MODEL_POSE='3,3,0.5,0,0,0,0,0' && 
 cd ~/PX4-Autopilot && make px4_sitl gz_x500_vision 
```
Error:
```
INFO [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
INFO [init] found model autostart file as SYS_AUTOSTART=4005
INFO [param] selected parameter default file parameters.bson
INFO [param] importing from 'parameters.bson'
INFO [parameters] BSON document size 316 bytes, decoded 316 bytes (INT32:13, FLOAT:3)
INFO [param] selected parameter backup file parameters_backup. bson
INFO [dataman] data manager file './dataman' size is 7872608 bytes
INFO [init] Gazebo simulator
INFO [init] gazebo already running world: aruco_world
WARN [init] PX4_GZ_MODEL_POSE not set, spawning at origin.
INFO [gz_bridge] world: aruco_world, model name: x500_vision_0, simulation model: x500_vision
INFO [gz_bridge] Requested Model Position: 0,0,0,0,0,0,0,0,0,0
INFO [gz_bridge] Model position z is less than or equal to 0. 0, moving upwards
ERROR [gz_bridge] timed out waiting for clock message
ERROR [gz_bridge] Task start failed (-1)
ERROR [init] gz_bridge failed to start and spawn model
ERROR [px4] Startup script returned with return value: 256
FAILED: src/modules/simulation/gz_bridge/CMakeFiles/gz_x500_vision /home/vor/PX4-Autopilot/build/px4_sitl_default/src/modules/simulation/gz_bridge/CMakeFiles/gz_x500_vision 
cd /home/vor/PX4- Autopilot/build/px4_sitl_default/src/modules/simulation/gz_bridge && /usr/bin/cmake -E env PX4_SIM_MODEL=gz_x500_vision /home/vor/PX4-Autopilot/build/px4_sitl_default/bin/px4
ninja: build stopped: subcommand failed.
make: *** [Makefile:232: px4_sitl] Error 1
```

Translated with [DeepL.com](https://www.deepl.com/?utm_campaign=product&utm_source=web_translator&utm_medium=web&utm_content=copy_free_translation) (free version)

Jaeyoung-Lim, 2025-07-24:
@Meowchine1 You should not run the gz sim before running the make target.

If you still have issues, please post a new issue, rather than hijacking this thread.