PX4/PX4-Autopilot #22159: [Bug] FW_LND_THRTC_SC is only displaying Integer Values

jstrebel, 2023-09-29:
### Describe the bug

It I try to enter for example a value of 0.7, I get in the QGC valuefield the value of 1 displayed.
Entering 1.6 I get 2

### To Reproduce

drohne switched on

### Expected behavior

Show the values in real format 0.5 displayed as 0.5

### Flight Log

https://review.px4.io/plot_app?log=6935219b-1314-4011-a257-37d81fded574

### Software Version

1.14 rc1

### Flight controller

holybro mini

### Vehicle type

None

ryanjAA, 2023-09-30:
Which QGC version? I had seems this awhile ago. It’s a QGC rounding issue. >x.5 rounds up. Download your Params and see. firmware should be using and saving what you enter.

jstrebel, 2023-10-04:
4.2.8

sfuhrer, 2023-12-15:
Thanks for the report, should be fixed with https://github.com/PX4/PX4-Autopilot/pull/22554

closed 2023-12-16 by MaEtUgR