PX4/PX4-Autopilot #22160: [Bug] FMUv6XRT hardfault

dk7xe, 2023-09-29:
### Describe the bug

During testflying FMUv6XRT today i had two hardfaults that seems to be SDcard related.
The first one happened when downloading logfiles via USB.
The second one happened midair during a mission execution resulting in crash and loss of the test vehicle
see also https://github.com/PX4/PX4-Autopilot/pull/21321#issuecomment-1740832559

### To Reproduce

run fmurt1176 with px4/pr-imxrt1170-imxrt1062 or NXPHoverGames/PX4-Autopilot/pr-imxrt1170_uwb_precland

### Expected behavior

no hardfault at all

### Screenshot / Media

[image]

### Flight Log

[fault_2023_09_29_08_56_07.log](https://github.com/PX4/PX4-Autopilot/files/12771208/fault_2023_09_29_08_56_07.log)
[fault_2023_09_29_11_59_28.log](https://github.com/PX4/PX4-Autopilot/files/12771218/fault_2023_09_29_11_59_28.log)
 https://logs.px4.io/plot_app?log=4fb99f3a-2c14-4fea-a653-5f9d2f47c675

### Software Version

Software Version:	[9a94ea3a](https://github.com/PX4/Firmware/commit/9a94ea3aeade1227d222ba468b29321dfe52af77)
branch: pr-imxrt1170_uwb_precland

### Flight controller

NXP fmurt1176

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

NXP fmurt1176 has FETtec ESC 45A 4in1 connected via PWM. T1 ethernet port of FMU connected to NXP T1ETH8 switch.
navq companion connected via 1000base-TX to T1ETH8 switch. Internal Mavlink comm via ethernet.
Pixhawk4 GPS on GPS1

### Additional context

to be assigned to @david

dk7xe, 2023-09-29:
@PetervdPerk-NXP @igalloway @mrpollo fyi

davids5, 2023-11-01:
@dk7xe - Please retest with the[ latest code](https://github.com/PX4/PX4-Autopilot/pull/22263) and see if this is still an issue.

dk7xe, 2023-11-11:
No issue any longer. Can be closed. Thank you!

closed 2023-11-13 by davids5