PX4/PX4-Autopilot #22163: [Bug] "Configuring GPS as Yaw/Heading Source" not working

MuhammadBilal1, 2023-10-01:
### Describe the bug

I am using [H-RTK Unicore UM982 (Dual Antenna)](https://holybro.com/products/h-rtk-unicore-um982) _GPS as Yaw/Heading Source_, but it's not working. I am unable to "ARM" the drone after switching to position mode. If I change the mode to "Altitude", the drone "ARM" successfully and I can fly that.

Issues:

- Unable to set the value of “EKF2_GPS_CTRL” to “Dual Antenna Heading” (8).
- Unable to arm the drone while in “Position” mode, upon pressing the “arm” button from remote, the GPS red light becomes flashing red in fast and a beep came. No error on QGC screen. Drone is arming if I arm it in the “Altitude” flight mode.

I can successfully set the value of the below parameters as asked in the [documentation](https://docs.px4.io/main/en/gps_compass/rtk_gps_holybro_unicore_um982.html#px4-configuration), but getting a message while trying to set the value of “EKF2_GPS_CTRL”.

- [SER_GPS1_BAUD](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#SER_GPS1_BAUD) → 230400
- [GPS_1_PROTOCOL](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#GPS_1_PROTOCOL) → 6: NMEA
- [GPS_YAW_OFFSET 1](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#GPS_YAW_OFFSET) → 90

Whenever, I try to set the value of “EKF2_GPS_CTRL” to “8” (Dual Antenna Heading), the value automatically goes back to “15”.

### To Reproduce

- Power ON the drone
- Power ON the Remote
- Make sure the flight mode is set to "Position".
- ARM the drone using the Remote or using QGC.

### Expected behavior

As far as I learned, by using the **UM982 Dual Antenna GPS**, I will use the UM982 for heading (to overcome the EMF issue, drone isn't doing YAW due to EMF).

### Screenshot / Media

**Error I am getting upon setting the value of "EKF2_GPS_CTRL" to 8.**
[image]

**My setup (this is the blueprint for testing, actual setup is done over the drone)**
[image]

### Flight Log

Initially, the flight lot wasn't showing anything, actually the log file wasn't being generated because the log type was set to log only after "ARM", but then I changed the values of **SDLOG_*** to the following:

- SDLOG_MODE --> 2
- SDLOG_PROFILE --> 1835

https://review.px4.io/plot_app?log=1e6deff0-c44b-4952-8c08-9369df9b28a9

### Software Version

**PX4 Version:** [1.14.0 rc](https://github.com/PX4/PX4-Autopilot/releases/tag/v1.14.0-rc1)

**Airframe:** Quadrotor x (Generic Quadcopter)

### Flight controller

Pixhawk 6C

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

I am using **Pixhawk 6C**, using **UM982** as Dual Antenna GPS (for heading source) at **GPS 1** port.

MuhammadBilal1, 2023-10-01:
@julianoes  @JonasVautherin  : Can you please help me in the above said issue! I am stuck here. :-(
Thank you.

bresch, 2023-10-02:
@MuhammadBilal1 You need to change `EKF2_HGT_REF` to something else than "GPS" to be allowed to deselect "Altitude" in `EKF2_GPS_CTRL`

MuhammadBilal1, 2023-10-04:
@bresch  @LorenzMeier Can you please help me here.
Thank you.

julianoes, 2023-10-05:
@MuhammadBilal1 isn't EKF2_GPS_CTRL set to 15 correct? The default is 7 and when you add 8, you get to 15.

MuhammadBilal1, 2023-10-05:
> @MuhammadBilal1 isn't EKF2_GPS_CTRL set to 15 correct? The default is 7 and when you add 8, you get to 15.

So **15** is the correct value if I want to use the "Dual antenna heading" feature of "UM982" GPS?

julianoes, 2023-10-05:
The checkbox adds bits. See https://docs.px4.io/main/en/advanced_config/parameter_reference.html#EKF2_GPS_CTRL

| Setting  | Bit  | Binary value
|---|---|---|
|  Lon/lat | 0  | 1 |
| Altitude  | 1  | 2 |
| 3D velocity  | 2  | 4 |
| Dual antenna heading | 3 | 8 |
| **Sum** | 0,1,2,3 | 15 |

MuhammadBilal1, 2023-10-06:
@julianoes Thank you for the detailed clarification.
Ok. I enabled the "Dual antenna heading" option, using UM928 with dual antenna (primary on nose & secondary on tail side).

**Is there a way to know that now my drone is using the yaw from the UM982 and not from the default compass!**
Here is the flight log where I am using UM982, but I am unsure if the yaw is being taken from the UM982.
https://review.px4.io/plot_app?log=84c84ab4-f444-4c1c-a930-b84c48c74cfc

Please note, I am setting the "Internal Magnetometer" to "OFF" because of the greater value of the EMF around my Pixhawk.

julianoes, 2023-10-06:
@MuhammadBilal1 check the estimator_status topic, specifically the control_mode_flags:

https://github.com/PX4/PX4-Autopilot/blob/aa87b2ef4dfa4009449b8ee4df595d40a51f4655/msg/EstimatorStatus.msg#L42

bresch, 2023-11-13:
> Here is the flight log where I am using UM982, but I am unsure if the yaw is being taken from the UM982

There is no GNSS heading data in that log
[image]
Fix type is "3D", that not sufficient to provide heading.

MuhammadBilal1, 2023-11-13:
> > Here is the flight log where I am using UM982, but I am unsure if the yaw is being taken from the UM982
> 
> There is no GNSS heading data in that log [image] Fix type is "3D", that not sufficient to provide heading.

@bresch Thank you for taking a look into this.
I discussed this issue with Holybro technical support and explored this issue in more depth.
But, at the end they asked me to create a post at https://discuss.px4.io, so I created the post along with the maximum details:
Please have a look here ---> https://discuss.px4.io/t/um982-isnt-providing-heading-data/35195 

It's not just the PX4 end, but I am not getting data over Uprecise tool of UM982.

bresch, 2023-11-13:
Alright, I hope that you'll find the root cause of that issue with the help of Holybro. I'll close this issue since it's not directly a PX4 bug.

closed 2023-11-13 by bresch