PX4/PX4-Autopilot #22180: [Bug] Gazebo SITL does not work if Gazebo Harmonic is installed

jmackay2, 2023-10-05:
### Describe the bug

If Gazebo Harmonic is installed, the simulation starts and Gazebo shows up, but the aircraft will not take off due to preflight check failures.

Error:
```
WARN  [health_and_arming_checks] Preflight Fail: ekf2 missing data
```

Full startup log:
```
[919/920] cd /autopilot/build/px4_sitl_default/src/modules/simulation/gz_bridge && /usr/bin/cmake -E env PX4_SIM_MODEL=gz_x500 /autopilot/build/px4_sitl_default/bin/px4

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=4001
INFO  [param] selected parameter default file parameters.bson
INFO  [param] importing from 'parameters.bson'
INFO  [parameters] BSON document size 274 bytes, decoded 274 bytes (INT32:11, FLOAT:3)
INFO  [param] selected parameter backup file parameters_backup.bson
INFO  [dataman] data manager file './dataman' size is 7866680 bytes
INFO  [init] starting gazebo with world: /autopilot/Tools/simulation/gz/worlds/default.sdf
WARN  [init] PX4_GZ_MODEL_NAME or PX4_GZ_MODEL not set using PX4_SIM_MODEL.
INFO  [gz_bridge] world: default, model name: x500_0, simulation model: x500
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
MESA: error: Failed to query drm device.
libGL error: glx: failed to create dri3 screen
libGL error: failed to load driver: iris
libGL error: failed to open /dev/dri/card0: No such file or directory
libGL error: failed to load driver: iris
Unknown message type [8].
Unknown message type [8].
Unknown message type [9].
Unknown message type [9].
Unknown message type [9].
Unknown message type [9].
Unknown message type [9].
INFO  [lockstep_scheduler] setting initial absolute time to 8000 us
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [tone_alarm] home set
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2023-10-05/02_58_38.ulg	
INFO  [logger] Opened full log file: ./log/2023-10-05/02_58_38.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [tone_alarm] notify negative
WARN  [health_and_arming_checks] Preflight Fail: ekf2 missing data
```

### To Reproduce

Tested in Docker to verify the issue happens only after Gazebo Harmonic is installed.

1. Start the docker:
```
docker run -it --rm -v /tmp/.X11-unix:/tmp/.X11-unix:ro -e DISPLAY=$DISPLAY -v /home/user/PX4-Autopilot:/autopilot px4io/px4-dev-ros2-rolling
```

2. Install gazebo in the docker:
```
apt update
apt dist-upgrade -y
apt get install lsb-release wget gnupg -y
wget https://packages.osrfoundation.org/gazebo.gpg -O /usr/share/keyrings/pkgs-osrf-archive-keyring.gpg
echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/pkgs-osrf-archive-keyring.gpg] http://packages.osrfoundation.org/gazebo/ubuntu-stable $(lsb_release -cs) main" | sudo tee /etc/apt/sources.list.d/gazebo-stable.list > /dev/null
apt update
apt install gz-garden
```

3. Test with Gazebo Garden, which works just fine:
```
cd autopilot
make px4_sitl gz_x500
```
This should output a happy message like:
```
pxh> INFO  [commander] Ready for takeoff!
```

4. Install Gazebo Harmonic
```
apt install gz-harmonic
```

5. Clean and rebuild, which will show the failure:
```
make clean
make px4_sitl gz_x500
```
This should show the failure:
```
WARN  [health_and_arming_checks] Preflight Fail: ekf2 missing data
```

### Expected behavior

I expected it to work with Gazebo, whether by still building with Garden, which is still available, or working with Harmonic. 

### Flight Log

SITL

### Software Version

`main` branch. 

### Flight controller

SITL

### Vehicle type

None

zyy37, 2023-10-09:
Why doesn't even my gazebo garden work? [link](https://github.com/PX4/PX4-Autopilot/issues/22121)

closed 2023-10-11 by dagar

anhducad1111, 2024-11-20:
got the same error