PX4/PX4-Autopilot #22191: [Bug] Fixed-wing simulation - Landing aborted: terrain measurement not found

alelse, 2023-10-06:
### Describe the bug

When flying a mission in SITL using the plane_lidar model and Gazebo classic simulator, a "Landing aborted: terrain measurement not found" message occurs when the plane model enters the Landing pattern stage of a mission.
The plane aborts and enters a loiter and does not land.

Set up uses the default parameters of the Gazebo classic plane model with Lidar -  px4_sitl gazebo-classic_plane_lidar

### To Reproduce

Steps to reproduce:
1. run 'make px4_sitl gazebo-classic_plane_lidar'
2. run QGroundControl
3. Create simple mission: - Takeoff,  Waypoint, Landing Pattern
4. Upload mission
5. Start mission
6. See error  "Landing aborted: terrain measurement not found" when plane enters Landing pattern
 
    

### Expected behavior

Landing pattern is completed successfully with the plane landing safely on ground which no warnings.

### Screenshot / Media

[image]

### Flight Log

https://logs.px4.io/plot_app?log=2f5a9374-4ffb-4f35-ae26-50da4d3db769

### Software Version

Output from MAVLink Console ver all command: 
HW arch: PX4_SITL
PX4 git-hash: 38b0de94c1b53394f8550e331c081395833ddbdd
PX4 version: 1.14.0 80 (17694848)
PX4 git-branch: main
OS: Linux
OS version: Release 5.15.90 (84892415)
Build datetime: Sep  6 2023 11:07:38
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.4.0
PX4GUID: 1006554d49534954414c44494e4f30303030
UNKNOWN MCU

 

### Flight controller

QGroundController

### Vehicle type

Fixed Wing

tstastny, 2023-10-06:
@Jaeyoung-Lim is the gazebo classic plane with lidar actually working (with lidar)? 

@alelse - otherwise, please see the docs here on params that you can set for automatic abort behavior https://docs.px4.io/main/en/flight_modes/mission.html#automatic-abort

Jaeyoung-Lim, 2023-10-06:
> @Jaeyoung-Lim is the gazebo classic plane with lidar actually working (with lidar)?

As far as I know, this was how the automatic landing was being tested no?

tstastny, 2023-10-06:
hm. distance sensor looks weird
[image]

frede791, 2023-10-07:
I think this is not particular to plane_lidar; I have encountered a similar issue with the regular plane in new Gazebo.

Jaeyoung-Lim, 2023-10-08:
@frede791 This is expected for the regular plane, since it does not have a lidar

Jaeyoung-Lim, 2023-10-09:
@tstastny The asphalt plane is the only collision geometry in the world(to save compute), so you will only see distance measurements right before the landing. We can make it bigger if you think it is necessary.

tstastny, 2023-10-09:
Ah ok. I guess it's a bit hard to handle any case someone simulates with a one size fits all gazebo map (or is infinite asphalt a possibility?). I would say as long as the sitl tests have the necessary runway to check behavior, maybe we can just make a note on docs, eg FAQ for plane sim to warn the ground will disappear?

Jaeyoung-Lim, 2023-10-09:
>  (or is infinite asphalt a possibility?).

Not really, but we can have a "bigger" asphalt plane if needed. 

> I would say as long as the sitl tests have the necessary runway to check behavior, maybe we can just make a note on docs, eg FAQ for plane sim to warn the ground will disappear?

Does this mean that landing works in SITL? Or was this actually never tested?

Jaeyoung-Lim, 2023-10-09:
@tstastny  I looked a bit more into the problem, and the real reason for this is that the land abort altitude `MIS_LND_ABRT_ALT`=30m for the default model, but the laser maximum range finder of the gazebo model is 15m.

What would be a realistic value for a default setup? After discussing with @sfuhrer, I think we want to make it "default" that the user is required to install a laser range finder on a fixed-wing vehicle. However, I am not sure what the expectation would be for the performance of the distance sensor.

tstastny, 2023-10-10:
60m is probably the default max trustable range (think lidar lite), but there are now several newer point lidars available that you could probably say 100m is trustable as well.

Jaeyoung-Lim, 2023-10-10:
> 60m is probably the default max trustable range (think lidar lite), but there are now several newer point lidars available that you could probably say 100m is trustable as well.
> 
> 30m still seems like reasonable default to me for abort alt. But of course a user needs to know the specs of the laser they use. (We obviously didn't do our due diligence in the sitl config in this case)

I am more asking what the user is expected to install as default (cheapest option) on their fixed wings, since we are now "requiring" lidar sensors on a fixed wing by default and flying without a lidar sensor is the exception. I checked the datasheet and 40m seems to be the max range for lidar lite v3. Should I set the laser range to 40m in this case?

tstastny, 2023-10-10:
> 40m seems to be the max range for lidar lite v3

oo - am i misremembering this sense too fondly? :S 

sure then - let's put 40m for sim. and for real fixed-wings.. yeah honestly people need to put lidars on their planes. i think this is reasonable to have as default expectation.

ryanjAA, 2023-10-10:
> I am more asking what the user is expected to install as default (cheapest option) on their fixed wings, since we are now "requiring" lidar sensors on a fixed wing by default and flying without a [...]

Anything less than 15m is dicey.... Older Lidar lite has practical range of 14m. Newer is probably 20-25m (real world).
Lightware units are 40-50% (sometimes 60) of stated range. So 100m option works at 40-60m and the 200m is >

As for testing in SITL - all works fine but you need to run plane lidar and then I went in and increased the lidar sim to 100m... 

Docs probably need an update and then as you both mentioned an increased value for sim. In real-world, auto landing with loiter to alt <12m is not ideal. It is probably fine but you start getting close to the ground for any type of deviation, wind, etc... Could it do it, probably but I like to cap at 12m. Just thinking about coming from say 250m and a slight overshoot on alt. At 20m, it is fine but at a 10m loiter, it is a bit close...

Jaeyoung-Lim, 2023-10-11:
@sfuhrer @ryanjAA @tstastny Putting the ignorant "getting started" PX4 user hat here. Can we agree on a number that is reasonable for a person to put on a fixed-wing drone that is being built for the first time? I think the default behavior of the SITL vehicle should be based on this.

jnomikos, 2023-11-10:
Ran into the same issue here with regular plane. Since it does not have lidar, I now understand why I am running into it.

sfuhrer, 2023-11-13:
> Can we agree on a number that is reasonable for a person to put on a fixed-wing drone that is being built for the first time? I think the default behavior of the SITL vehicle should be based on this.

Not sure if we have to come up with a range requirement. Some users may want to use a 10m distance sensor, and if they plan the landing accordingly it's fine. I would instead make it clear in the docs that **you have to plan the loiter down waypoint no higher than the range of your distance sensor.**
@tstastny wdyt?

Jaeyoung-Lim, 2023-11-13:
@sfuhrer Was more asking what range value I should set for SITL that is representative

sfuhrer, 2023-11-13:
> @sfuhrer Was more asking what range value I should set for SITL that is representative

Ah okay - I propose 30m for that.

ryanjAA, 2023-11-13:
30m is fine. I have it set to 100m in ours but 30-50 is actually better to learn what happens if you are too high and measurement isn't found and auto abort, etc etc

Jaeyoung-Lim, 2023-11-13:
> Ah okay - I propose 30m for that.

And which laser range finder does this correspond to? (Basically what should people be putting on to FW as default sensor configuration)

hamishwillee, 2023-11-15:
> sure then - let's put 40m for sim. and for real fixed-wings.. yeah honestly people need to put lidars on their planes. i think this is reasonable to have as default expectation.

@sfuhrer @tstastny If there is an expectation that an FW vehicle will have a distance sensor then that needs to appear in the assembly guide and the configuration guide. 
- Is this the case?
- Do we have a recommendation on the lidar?
- Is there any specific tuning required?
- If it is required and is not present, what will the user see? Can we detect the absence of this sensor and warn or auto-config the behaviour away?

hamishwillee, 2023-11-15:
Lastly, you guys clearly know what world to use for different use cases. It might be useful to list a bunch of cases we think people might need and the simulations that map to those.

sfuhrer, 2023-11-15:
> And which laser range finder does this correspond to

We almost exclusively use the Lightware LW20/C. Has (effective) range well above what you ever need for a fixed-wing landing (50-80m).

sfuhrer, 2023-11-15:
> If it is required and is not present, what will the user see? Can we detect the absence of this sensor and warn or auto-config the behaviour away?

We currently only mention it in the docs that using a distance sensor is strongly recommended. Maybe we should even throw by default pre-flight errors if none is connected? We could do that by setting SYS_HAS_NUM_DIST to 1 in the generic plane.

ryanjAA, 2023-11-15:
Preflight check would be great. 

other 100m lightware units are extremely common I’d say too, sf11, etc. same range, different external package only.

gurselturkeri, 2024-03-04:
> @tstastny I looked a bit more into the problem, and the real reason for this is that the land abort altitude `MIS_LND_ABRT_ALT`=30m for the default model, but the laser maximum range finder of the gazebo model is 15m (https://github.com/PX4/PX4-SITL_gazebo-classic/blob/20ded0757b4f2cb362833538716caf1e938b162a/models/lidar/model.sdf#L53). Therefore, the landing is aborted before the terrain is within range of the distance sensor.
> 
> What would be a realistic value for a default setup? After discussing with @sfuhrer, I think we want to make it "default" that the user is required to install a laser range finder on a fixed-wing vehicle. However, then IMHO we need to be clear on how powerful of a distance sensor is expected so that the fw landing works. We should adjust the mission landing abort accordingly.

Hi, I am using gazebo and I got a same error for fixed wing (advanced_plane) which has not lidar sensor. How can I solve that problem. Thank you.

Jaeyoung-Lim, 2024-03-04:
@gurselturkeri It is the correct behavior that the landing is aborted when you don't have a lidar sensor.

If you want to test autolanding, you need to add the lidar sensor to the model OR use a model that already has one.

gurselturkeri, 2024-03-04:
> @gurselturkeri It is the correct behavior that the landing is aborted when you don't have a lidar sensor.
> 
> If you want to test autolanding, you need to add the lidar sensor to the model OR use a model that already has one.

I have changed FW_LND_USETER value to 0 now it is working fine.

Pan-Chenliang, 2024-09-10:
Is there a method to add a lidar on rc_cessna in gazebo garden? It seems those lidar model can only be used in gazebo-classic

DonLakeFlyer, 2025-02-28:
The ability for me to test landing patterns in QGC with SITL is kind of important. For example to test things like this; https://github.com/mavlink/qgroundcontrol/pull/12313#issuecomment-2678078402. I'm using new Gazebo but there aren't any planes which have LIDAR hence I can't test landing. Is it possible to add a plane with lidar so I can test?

Jaeyoung-Lim, 2025-02-28:
@DonLakeFlyer Use `plane_lidar`

DonLakeFlyer, 2025-02-28:
As far as I can tell the new Gazebo does not have a plan_lidar. I think only classic has that.
```
parallels@ubuntu-linux-2404:~/repos/PX4-Autopilot$ HEADLESS=1 make px4_sitl plane_lidar
ninja: error: unknown target 'plane_lidar'
make: *** [Makefile:232: px4_sitl] Error 1
```

And I can't get classic to run for me on my Linux arm vm.

dakejahl, 2025-03-01:
@DonLakeFlyer give this a try https://github.com/PX4/PX4-gazebo-models/pull/82

If there's anything else needed let's discuss it in that PR