PX4/PX4-Autopilot #22193: [Bug] What's the reason of drone crashed? (PX4 1.14.0 rc, Pixhawk 6C, QuadOcta, UM982)

MuhammadBilal1, 2023-10-06:
### Describe the bug

Yesterday, I completed the update of my QuadOcta drone from **Pixhawk 4** to **Pixhawk 6C** (PX4 1.14.0-rc) along with the UM982 Dual heading antenna GPS.

In my initial migrations, I updated the Pixhawk, GPS, Actuator Controls, imported the PID tuning parameters from my previous flights.

In my first flight, I felt my drone isn't much stable, so then I decided to perform Autotune (it was new frame).
My auto-tune went well and parameters saved successfuly.

After auto-tune, I started a new mission flight and my drone crashed after  5-8 seconds of takeoff.

Can you please review the log and share some details about the reason of the crash.

### To Reproduce

Mission flight

### Expected behavior

I was expecting a more stable flight than earlier because I perform a round of auto-tune. But, it went opposite and my drone crashed.

### Flight Log

**Setup:** Pixhawk 6C, UM982 Dual antenna heading

**1-** First flight after migration from Pixhawk 4 to Pixhawk 6C (PX4 1.14.0-rc)(Internal Magnetometer was OFF): 
https://review.px4.io/plot_app?log=0d3d9e8a-6f76-4586-84c8-07d19e397de4

**2-** Auto-tune flight:
https://review.px4.io/plot_app?log=2bcae38b-6a3d-4439-94ec-7d697b06838e

**3-** Mission flight after performing auto-tune:
https://review.px4.io/plot_app?log=df00d3ed-5412-454b-a4d8-58bafdddc556

### Software Version

PX4 - 1.14.0-rc

### Flight controller

Pixhawk 6C

### Vehicle type

Multicopter

eogks8921, 2023-10-12:
When checking the flight data, it appears that the roll rate I gain and pitch rate I gain are high. Additionally, it seems that the roll rate P gain and pitch rate P gain are low.

Looking at the actuator control graph, it can be observed that the hover flight output is above 0.8, indicating a lack of thrust relative to the drone's weight. This implies insufficient control margin and may lead to a degradation in flight performance.

Lastly, examining the magnetic field data reveals that as thrust increases, the magnetic norm value changes, likely leading to yaw drift. It might be advisable to adjust the GPS position (avoiding power sources) to address this issue.

Gnaneswar484, 2023-10-19:
Trying to fly the drone to desired position using code in qgc and jmavsim.

I want to move drone to particular position using the code below mentioned, I build the code successfully into px4-firmware and when running the code using jmavsim it was successful, but the drone was not moving. Can anyone of you check is there any problem with the code as I am new to the px4-development, I am not able to debug it.

#include <px4_platform_common/px4_config.h>
#include <px4_platform_common/tasks.h>
#include <uORB/uORB.h>
#include <uORB/topics/vehicle_local_position_setpoint.h>
#include <uORB/topics/vehicle_local_position.h>

extern "C" __EXPORT int my_autonomous_flight_module_main(int argc, char *argv[]);

static volatile sig_atomic_t _should_exit = false;
static int _task = -1;

// Target position (latitude, longitude, altitude)
static constexpr double target_pos[] = {47.398036, 8.545484, 5.0}; // Replace with your target position

int my_autonomous_flight_module_thread(int argc, char *argv[]) {
PX4_INFO("My Autonomous Flight Module is running!");

orb_advert_t pos_sp_adv = orb_advertise(ORB_ID(vehicle_local_position_setpoint), nullptr);

// Main loop
while (!_should_exit) {
    // Create a new setpoint
    struct vehicle_local_position_setpoint_s sp{};
    sp.x = target_pos[0];
    sp.y = target_pos[1];
    sp.z = target_pos[2];

    // Publish the setpoint
    orb_publish(ORB_ID(vehicle_local_position_setpoint), pos_sp_adv, &sp);

    px4_usleep(1000000); // Sleep for 1 second
}

PX4_INFO("My Autonomous Flight Module exiting.");
return 0;
}

extern "C" int my_autonomous_flight_module_main(int argc, char *argv[]) {
// Start the module's main thread
_task = px4_task_spawn_cmd(
"my_autonomous_flight_module",
SCHED_DEFAULT,
SCHED_PRIORITY_DEFAULT + 40,
2000,
my_autonomous_flight_module_thread,
(char *const *)nullptr);

if (_task < 0) {
    PX4_ERR("Task start failed");
    return -errno;
}

return 0;
}