PX4/PX4-Autopilot #22197: [Bug] accelerometer_integral_dt is vibration.

Abner0907, 2023-10-08:
### Describe the bug

My code is base on the version of v1.12.3.
I have done some changes in the PX4-Autopilot. But I have not change drivers about the ICM20689.

now I found sometimes the `accelerometer_integral_dt` and `gyro_integral_dt` in the topic of `sensor_comibined` has vibrated as the picture below(the normal is near the 5000, but now it can go to 1250 or 46250 randomly with the same code): show as below: 
my log is : 
https://logs.px4.io/plot_app?log=3299a2a2-f7de-433c-bbf8-c306fb9d8c92
https://logs.px4.io/plot_app?log=c2d89a54-b67d-4e69-827e-ae42856ab1d4.
I am very appreciate that who can give me some help.

@dagar @bresch Do you have the similar issue? I found that you have found a issue here https://github.com/PX4/PX4-Autopilot/issues/17804.

[image]
[image]

### To Reproduce

See error

### Expected behavior

nothing

### Flight Log

https://logs.px4.io/plot_app?log=c2d89a54-b67d-4e69-827e-ae42856ab1d4
https://logs.px4.io/plot_app?log=3299a2a2-f7de-433c-bbf8-c306fb9d8c92

### Software Version

v1.12.3

### Flight controller

Multicopter

### Vehicle type

Multicopter

dagar, 2023-10-08:
> @dagar @bresch Do you have the similar issue? I found that you have found an issue here https://github.com/PX4/PX4-Autopilot/issues/17804.

No, depending on the board there could be a little bit of jitter due to things like multiple sensors sharing the same bus, but even that should be on the order of tens to hundreds of microseconds (eg ~1% of that integration period). Something is very wrong if you're seeing it jump to ~46 milliseconds periodically.

I'm not sure we'll be able to help all that much because this is custom hardware on older PX4, but with everything running you could grab the output of `sensors status`, `work_queue status`,`uorb top`, `top`, and maybe `perf`. From that we might be able to find the culprit.

Abner0907, 2023-10-09:
@dagar  thank you for your reply.
I share the outputs as below. 
onething I find the different, the normal is show as below: 
[image]

the below show the unnormal state: 
[image]
[image]
[image]
[image]
[image]

Abner0907, 2023-10-10:
I doubt the reason is caused by the driver of `ICM20689` or  `vehicle_imu`  or `sensors` thread. 
just record  the relates PR and issues with the similar issue for improvement.
https://github.com/PX4/PX4-Autopilot/issues/17804  (variations in sensor_combined/accelerometer_integral_dt )
https://github.com/PX4/PX4-Autopilot/pull/17805 (drivers/imu/invensense: store offset registers on configure, trigger reset on any change)
https://github.com/PX4/PX4-Autopilot/pull/17817 (sensors/vehicle_imu: minor IMU integration improvements)
https://github.com/PX4/PX4-Autopilot/pull/17818 (unmerge drivers/imu: minor scheduling improvements to tolerate occasional jitter (eg multiple sensors sharing same bus))
https://github.com/PX4/PX4-Autopilot/pull/17862 (drivers/imu: set sample timestamp from interrupt (if available))