PX4/PX4-Autopilot #22219: [Bug] Broken px4vision model in Gazebo.

frede791, 2023-10-13:
### Describe the bug

The px4vision model does not work on Gazebo. It opens gazebo, but no model is loaded. Instead the error as shown below in the screenshot pops up. It appears to be segmentation fault. This does not happen with any other tested vehicles (standard_vtol, rc_cessna, x500 variants).

### To Reproduce

Running inside the PX4-Autopilot:
`make px4_sitl gz_px4vision`

### Expected behavior

Model should load into gazebo.  

### Screenshot / Media

[image]

### Flight Log

-

### Software Version

Using newest commit: 476b5d5594f6de867fcf2d59f5d56ffb1a2dc4d0

### Flight controller

It's in simulation.

### Vehicle type

None

Jaeyoung-Lim, 2023-10-13:
@frede791 Thanks for raising this, I can confirm that this is a problem

frede791, 2023-10-13:
@Jaeyoung-Lim The xx-inertia values on the rotors seem way too large to me? If you place a minus (so 10^-7 rather than 10^7), I can load the model, but takeoff still does not work.

Claudio-Chies, 2024-07-15:
`make px4_sitl gz_px4vision` seems to work now, so i'll close this for now.

closed 2024-07-15 by Claudio-Chies