PX4/PX4-Autopilot #22220: Holybro UART optical flow (thoneflow driver) severe oscillations position mode [Bug]

dirksavage88, 2023-10-13:
### Describe the bug

I've tested the holybro uart sensor on a 210mm wheelbase quad and found severe position oscillations when using 1.14, main, and even going back to 1.13.3 stable. I swapped the holybro sensor for the Ark can flow sensor with updated cannode firmware (recent main) that includes the gyro bias estimation fixes. The ark flow worked great.

See this forum discussion:
https://discuss.px4.io/t/holybro-optical-flow-sensor-thoneflow-oscillations-on-main/34768

### To Reproduce

Enable the thoneflow driver, set the OF, EKF2, and rangefinder params, and take off in position mode

### Expected behavior

Optical flow position mode holds stable position with minimal oscillation

### Screenshot / Media

Here is a pitch angle plot from the holybro sensor: [image]

Same exact hardware setup except the ark flow: 
[image]

### Flight Log

Holybro (like a merry go round) : https://review.px4.io/plot_app?log=94031e37-9350-4b03-a76e-19fdd9cdcf4d
Ark can flow (I'm not moving): https://review.px4.io/plot_app?log=7ea5d915-144a-45ff-971c-76aa3d33732e 

### Software Version

tested with similar results on stable, 1.14 beta, recent main

### Flight controller

holybro 6c mini

### Vehicle type

None

dagar, 2023-10-18:
[image]

dirksavage88, 2023-10-20:
As requested @dagar @bresch here are three logs that show an attempt at a box pattern as well as spinning by hand in position mode while armed (but landed):

1) box pattern: https://review.px4.io/plot_app?log=08d11193-1a40-498f-b366-cd56d09862eb
2) Spinning 1: https://review.px4.io/plot_app?log=9ed53bbc-df6c-41e1-95c2-882b4cd0bfda
3) Spinning 2: https://review.px4.io/plot_app?log=6ad9c489-5c97-4fcc-b172-a645d395fc93
4) EKF2 replay from box pattern: https://review.px4.io/plot_app?log=27ad9e20-8aa6-4d33-a137-3d20e8fe3348

jfaure6, 2023-11-29:
Hello @dirksavage88 !

I observe exactly the same phenomenon with the same hardware (pmw3901, thoneflow driver in UART).
My setup is Holybro's px4vision in v1.5. Note that I also have the Holybro PX4Vision in v1, with the PMW3901 in SPI, but with this configuration, I do not observe any oscillations.
You can view the logs:
v1.5 :     https://review.px4.io/plot_app?log=d2530b83-3116-4b63-a292-8a3d440a3993
v1 :       https://review.px4.io/plot_app?log=e8908b3a-7745-41a9-859f-4bbdddd4349a

bresch, 2023-12-11:
Looks like the scale in the driver might be wrong. Try divide those scales by 2 (replace 3.52e-3 by 1.76e-3): https://github.com/PX4/PX4-Autopilot/blob/8aab3e80133e9f31b5826198309ba1c1d224302e/src/drivers/optical_flow/thoneflow/thoneflow_parser.cpp#L136-L137

mmalecki, 2024-01-05:
This patch has greatly improved OF performance with this sensor for me:

Before, note the oscillations with no RC input:
[image]

After, lack of earlier oscillations:
[image]

Thanks @bresch!

PR #22611 (2024-01-09): Fix scaling issue with thoneflow/holybro optical flow pmw3901 uart driver
Fixes #22220
closed 2024-01-09 by bresch