PX4/PX4-Autopilot #22221: [Bug] Terrain Hold: Local Z doesn't follow distance sensor

dakejahl, 2023-10-13:
### Describe the bug

I am testing a small 250 quad with an ARK Flow and I am noticing poor Z position hold. It appears from the data that the distance sensor measurements are not being used as the primary height source, even though I have `MPC_ALT_MODE=2` and the estimator status `show cs_rng_hgt=1`.

I would expect the Z local position to track the distance sensor readings. I did a second flight and reduced EKF2_RNG_NOISE from 0.1 to 0.05 and it made no difference. It's also evident it's using the barometer due to the ground effect when I try to hover near the ground.

I have EKF2_HGT_REF set to 0 (baro as primary height reference) but I would expect that the range finder would be used exclusively when EKF2_RNG_CTRL is enabled and conditions are met.

https://logs.px4.io/plot_app?log=ffa07b2c-fd47-4a0e-89c1-71f5d348ed2f
[image]

https://logs.px4.io/plot_app?log=e991014d-0b85-49ba-a3fa-ba109515e15b
[image]

### To Reproduce

Try to hover near the ground using Terran Hold

### Expected behavior

Perfect Z hold if distance sensor readings are good

### Flight Log

https://logs.px4.io/plot_app?log=e991014d-0b85-49ba-a3fa-ba109515e15b

### Software Version

main

### Flight controller

ark_fmu-v6x_default

### Vehicle type

Multicopter

julianoes, 2023-10-17:
@bresch is this expected?

bresch, 2023-10-18:
> I have EKF2_HGT_REF set to 0 (baro as primary height reference) but I would expect that the range finder would be used exclusively when EKF2_RNG_CTRL is enabled and conditions are met.

With this configuration, both sensors are used but the solution converges towards the barometer reading as this is the height reference. In order to switch the height reference to range when the conditions are met, you can set `EKF2_HGT_REF` to range finder and EKF2_RNG_CTRL to "conditional", see:
https://docs.px4.io/main/en/advanced_config/tuning_the_ecl_ekf.html#height

However, even with your current configuration, the `dist_bottom` is correctly estimated and is consistent with the range finder measurements, meaning that there is probably an issue with the terrain hold feature.
There is an open PR that most likely attempts to fix the same issue, but it's apparently unfinished: https://github.com/PX4/PX4-Autopilot/pull/21885 Maybe you could have a look at it.

dakejahl, 2023-10-18:
> With this configuration, both sensors are used but the solution converges towards the barometer reading as this is the height reference.

But not when EKF2_RNG_CTRL is conditonal and the conditions are met -- it _should_ use the range finder in this case. 

The solution in #21885 addresses an issue with the flight task setpoints, but this doesn't appear to be a problem with setpoints. The estimate tracks the setpoint, the problem is the estimate is not tracking the rangefinder
[image]

closed 2023-11-07 by dagar