PX4/PX4-Autopilot #22229: Holybro-KAKUTEH7-V1's magnetic sensor problem

kevinuav, 2023-10-17:
### Describe the bug

 I find the kakuteh7's magnetic sensor may have issue. I use kakuteh7-v1. It is installed in my fixed-wings. I connect an external magnetic sensor for it since the FC has no MAG inside. However. The estimated direction is almost reverse to the real. I have tried to correct it with the magnetic sensor setting. But it doesn't work. It should be a bug in the transplantation of the kakuteh7. 

### To Reproduce

When you connect the i2c external magnetic sensor for kakuteh7-v1. You will find its estimated direction is almost 160 degrees reverse to the real. And it can't be corrected with the magnetic installation parameters.

### Expected behavior

The estimated direction of the magnetic compass is almost 160 degrees reverses to the real when the vehicle keeps still (the GPS direction is invalid). This is critical for the plane's automatic takeoff.

### Flight Log

At about 2:15 to 2:18. You can see the magnetic estimated direction suddenly change to the real direction when my plane is taking off.(The GPS direction is valid).

https://logs.px4.io/plot_app?log=8405692c-0f61-46ee-834c-41338c26e5eb

### Software Version

V1.14.0-main

### Flight controller

Holybro-kakuteh7-v1

### Vehicle type

Fixed Wing

bresch, 2023-10-17:
The magnetometer data looks weird: it's almost a constant value with noise (https://logs.px4.io/plot_app?log=8405692c-0f61-46ee-834c-41338c26e5eb#Nav-Raw-Magnetic-Field-Strength). Are you sure it's working?
[image]

kevinuav, 2023-10-17:
> The magnetometer data looks weird: it's almost a constant value with noise (https://logs.px4.io/plot_app?log=8405692c-0f61-46ee-834c-41338c26e5eb#Nav-Raw-Magnetic-Field-Strength). Are you sure it's working?
> [image]

It can be calibrated successful. The FC can read the data from it.

julianoes, 2023-10-17:
@bresch us right, something is wrong with the data. I'll check with my board

julianoes, 2023-10-18:
So what I missed is that the KakuteH7 v1 does not have a mag included.

@kevinuav can you post a picture of the mag that you're using? I need to check if I have one that is the same.

kevinuav, 2023-10-18:
This is the data comes from the other plane of kakutah7-v1. It's estimated direction is almost 80 degrees different from the real during the taking off procedure at about 3:15-3:40. Until the GPS's direction becomes valid.

https://logs.px4.io/plot_app?log=c8350813-9f05-4eb7-8937-c60f849dbe14

kevinuav, 2023-10-18:
> So what I missed is that the KakuteH7 v1 does not have a mag included.
> 
> @kevinuav can you post a picture of the mag that you're using? I need to check if I have one that is the same.

OK, I will take a photo for it.

julianoes, 2023-10-18:
I don't think I have that one.

But it looks like it has the [qmc5883l](https://github.com/PX4/PX4-Autopilot/tree/e8a0a0772e2a4ae3490e55274ffd70ffdd7aac14/src/drivers/magnetometer/qmc5883l) sensor.

Can you try `qmc5883l status` in MAVLink shell?

kevinuav, 2023-10-18:
> I don't think I have that one.
> 
> But it looks like it has the [qmc5883l](https://github.com/PX4/PX4-Autopilot/tree/e8a0a0772e2a4ae3490e55274ffd70ffdd7aac14/src/drivers/magnetometer/qmc5883l) sensor.
> 
> Can you try `qmc5883l status` in MAVLink shell?

nsh> qmc5883l status
INFO  [SPI_I2C] Running on I2C Bus 1, Address 0x0D
qmc5883l: reset: 0 events
qmc5883l: bad register: 0 events
qmc5883l: bad transfer: 0 events
nsh> qmc5883l status

julianoes, 2023-10-18:
Nothing bad showing there. That's odd, and I'm afraid I can't reproduce it here because I don't have the hardware.

bresch, 2023-11-06:
> This is the data comes from the other plane of kakutah7-v1. It's estimated direction is almost 80 degrees different from the real during the taking off procedure at about 3:15-3:40. Until the GPS's direction becomes valid.
https://logs.px4.io/plot_app?log=c8350813-9f05-4eb7-8937-c60f849dbe14

[image]

In this log, the norm of the magnetic field is far from being constant (it should be a straight line with a bit of noise). Looking at the calibration parameters, the Y offset indicates that something produces a quite strong magnetic field near the magnetometer: `CAL_MAG0_YOFF, 1.576`

kevinuav, 2024-07-08:
Hi. I got a log which clearly shows the issue of the kakuteh7's magnetic sensor I mentioned before. The plane "side slips" completely after take off(5:39). Then it suddenly corrected at 5:58 itself.
Here is the log: https://review.px4.io/plot_app?log=730fe5ec-7059-477f-98b2-f96cc8ec942f
[image]