PX4/PX4-Autopilot #22245: [Bug] CAN Optical Flow PMW3901 [HereFlow] not subscribing

w-sribunma, 2023-10-19:
### Describe the bug

I have a Pixhawk 4 set up and flashed with PX4 v1.14.0-rc2. After running the "make px4_fmu-v5_default" command and flashing it onto the FC, the I/O indicator light rapidly blinks red, indicating an error.

We are trying to run PX4 with an optical flow sensor with position control mode without GPS. We are using a DroneCan PMW3901 Optical flow (HereFlow Optical Flow) sensor.  When checking the "dmesg," this message appears "ERROR [SPI_I2C] pmw3901: no instance started (no device on bus?)." It seems like flow reading is not being subscribed from the CAN port.

DroneCAN PMW3901 Optical Flow is not subscribed and the optical flow topic is not able to listen to optical flow from the CAN port

### To Reproduce

1. Connect PMW3901 OF (HereFlow Optical Flow Sensor / Lidar Module) sensor on "CAN1" port
2. run make px4_fmuv5_default on v1.14.0-rc2
3. on QGC, flash the Pixhawk board with the custom firmware
4. Set the following parameters
5. enable UAVCAN_SUB_FLOW
6. enable EKF2_OF_CTRL
7. enable SENS_EN_PMW3901
8. Recalibrate and reboot vehicle

### Expected behavior

Arming the vehicle shouldn't flash a red LED indicator. The vehicle should be able to take off and hover at an altitude and position with position control mode using optical flow without GPS. Optical flow topic should be subscribed via CAN port

### Screenshot / Media

[Error Log OF 10182023 - Google Docs.pdf](https://github.com/PX4/PX4-Autopilot/files/13037389/Error.Log.OF.10182023.-.Google.Docs.pdf)

### Flight Log

S500 model was not able to take off

### Software Version

PX4 Autopilot V1.14.0-rc2

### Flight controller

Pixhawk 4

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

PMW3901 OF sensor connected to CAN1 port, Range finder connected to UART & I2C B

Frame: S500

### Additional context

Replicating it on PX4 Vision drone v1.5 with Pixhawk 6C FC gives a similar error where the PMW3901 OF topic is not subscribing.

w-sribunma, 2023-10-19:
Here is the output. Thank you in advance!

uavcan status
Pool allocator status:
    Capacity hard/soft: 500/250 blocks
    Reserved:  62 blocks
    Allocated: 27 blocks

UAVCAN node status:
    Internal failures: 0
    Transfer errors:   0
    RX transfers:      4264
    TX transfers:      2737

CAN1 status:
    HW errors: 85
    IO errors: 91
    RX frames: 14589
    TX frames: 3061
CAN2 status:
    HW errors: 72335
    IO errors: 74759
    RX frames: 0
    TX frames: 628

ESC outputs:
INFO  [mixer_module] Param prefix: UAVCAN_EC
control latency: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
Channel Configuration:
Channel 0: func:   0, value: 0, failsafe: 65535, disarmed: 65535, min: 1, max: 8191
Channel 1: func:   0, value: 0, failsafe: 65535, disarmed: 65535, min: 1, max: 8191
Channel 2: func:   0, value: 0, failsafe: 65535, disarmed: 65535, min: 1, max: 8191
Channel 3: func:   0, value: 0, failsafe: 65535, disarmed: 65535, min: 1, max: 8191
Channel 4: func:   0, value: 0, failsafe: 65535, disarmed: 65535, min: 1, max: 8191
Channel 5: func:   0, value: 0, failsafe: 65535, disarmed: 65535, min: 1, max: 8191
Channel 6: func:   0, value: 0, failsafe: 65535, disarmed: 65535, min: 1, max: 8191
Channel 7: func:   0, value: 0, failsafe: 65535, disarmed: 65535, min: 1, max: 8191
Servo outputs:
INFO  [mixer_module] Param prefix: UAVCAN_SV
control latency: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
Channel Configuration:
Channel 0: func:   0, value: 0, failsafe: 500, disarmed: 500, min: 0, max: 1000
Channel 1: func:   0, value: 0, failsafe: 500, disarmed: 500, min: 0, max: 1000
Channel 2: func:   0, value: 0, failsafe: 500, disarmed: 500, min: 0, max: 1000
Channel 3: func:   0, value: 0, failsafe: 500, disarmed: 500, min: 0, max: 1000
Channel 4: func:   0, value: 0, failsafe: 500, disarmed: 500, min: 0, max: 1000
Channel 5: func:   0, value: 0, failsafe: 500, disarmed: 500, min: 0, max: 1000
Channel 6: func:   0, value: 0, failsafe: 500, disarmed: 500, min: 0, max: 1000
Channel 7: func:   0, value: 0, failsafe: 500, disarmed: 500, min: 0, max: 1000

Sensor 'flow':
name: uavcan_flow
channel 0: node id 124 --> instance 0

Sensor 'gnss':
name: uavcan_gnss

Sensor 'mag':
name: uavcan_mag

Sensor 'rangefinder':
name: uavcan_rangefinder
channel 0: node id 124 --> instance 0

Online nodes (Node ID, Health, Mode):
     124 OK         OPERAT    

uavcan: cycle time: 70998 events, 4088409us elapsed, 57.58us avg, min 13us max 58957us 304.687us rms
uavcan: cycle interval: 70998 events, 2378.98us avg, min 22us max 121118us 1277.556us rms
nsh>

dagar, 2023-10-19:
That looks like it's working, just don't set SENS_EN_PMW3901, you only need that if attaching it directly on SPI.

w-sribunma, 2023-10-20:
@dagar Ok, thank you! Do you happen to know what may be the reason for the rapid I/O port red LED flashing? It looks like an error, but I don't see any error related to the calibration process. I assume it is not related to optical flow because I cannot take off the vehicle even in manual or altitude mode.

closed 2023-10-23 by w-sribunma