PX4/PX4-Autopilot #22253: [Bug] QGC does not install latest stable 1.14 on Pixhawk 6X

mzahana, 2023-10-22:
### Describe the bug

I have QGC `4.2.8` (also tried the daily build), and Pixhaw 6X.
I tried to install the latest stable PX4 `v1.14.0`. However the reported version after installation is `v1.13.3`

I am not sure if this is PX4 bug or QGC bug.

### To Reproduce

1. Download latest stabel QGC
2. Flash the latest PX4 release (`v1.14.0` at the time of writing this issue)
3. QGC reports PX4 version as `v1.13.3`

### Expected behavior

QGC should report `v1.14.0`

### Flight Log

There is no need for flight log.

### Software Version

Supposed to be `v1.14.0`

### Flight controller

Pixhawk 6X

### Vehicle type

Multicopter

antonio-sc66, 2023-10-24:
Confirmed, happened to me too with FMU6X. I've also mentioned it on the discord platform. Looks like it's a QGC related error.

junwoo091400, 2023-11-06:
This means that https://px4-travis.s3.amazonaws.com/Firmware/stable/px4_fmu-v6x_default.px4 contains still the v1.13.3 binary. @dagar could you check Jenkins on why this wasn't updated?

MaEtUgR, 2023-11-07:
@dagar mentioned today in the maintainer call that there was a CI problem that delayed the final upload of 1.14.0 stable to ~27. October (after this issue). So it should all work by now but needs to be verified.

junwoo091400, 2023-11-07:
During the call it was discussed that the binary update was completed at October 27th.

And @beniaminopozzan verified that the current stable binary for v6x is up to date to 1.14. Could you check again?