PX4/PX4-Autopilot #22255: bug about check if all mission items are inside the geofence

czbxzm, 2023-10-23:
### Describe the bug

https://github.com/PX4/PX4-Autopilot/blob/f68f88b97c77d38054d54696fca5ef1a3e91822b/src/modules/navigator/mission_feasibility_checker.cpp#L117C3-L117C3
The above code will check if the mission waypoint is within the no-fly zone, this is to avoid the aircraft from flying over the no-fly zone while performing the mission. However, the mechanism only checks the waypoints and does not consider the case where the route crosses the no-fly zone.

### To Reproduce

1.draw the no fly zone and upload
2.Select two waypoints that are outside the no-fly zone and have the line cross the no-fly zone.
[image]

### Expected behavior

null

### Flight Log

null

### Software Version

1.14.0

### Flight controller

QGC

### Vehicle type

Multicopter

julianoes, 2023-10-23:
@czbxzm thanks for the feature request. Contributions are welcome!

sfuhrer, 2023-11-14:
That won't be easy to fix and check during the planning stage. But it already does work on execution, and will prevent the vehicle from entering the geofence.