PX4/PX4-Autopilot #22264: [Bug] QGC does not uploads the latest stable PX4 release  (v1.14.0) to mRo Control Zero H7

TOTON95, 2023-10-25:
### Describe the bug

I am running QGC `4.2.9` to upload the latest stable PX4 release `v1.14.0` into the mRo Control Zero H7 board, but it reports back that `v1.13.3` is installed instead (Summary section).

### To Reproduce

1. Install QGC v4.2.9.
2. Flash the latest stable PX4 release  (`v1.14.0`)  reported to QGC.
3. Once finished, observe the version in the Summary section.

### Expected behavior

- QGC should report `Firmware Version: 1.14.0` 

### Screenshot / Media

[image]
[image]

### Flight Log

N/A

### Software Version

```
HW arch: MRO_CTRL_ZERO_H7
FW git-hash: 1c8ab2a0d7db2d14a6f320ebd8766b5ffaea28fa
FW version: Release 1.13.3 (17630207)
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: 4a1dd8680cd29f51fb0fe66dcfbf6f69bec747cf
Build datetime: Mar 14 2023 01:29:46
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000600000000343538303332510a0036003f
MCU: STM32H7[4|5]xxx, rev. V
```

### Flight controller

mRo Control Zero H7

### Vehicle type

None

### How are the different components wired up (including port information)

N/A

### Additional context

N/A

ryanjAA, 2023-10-26:
Same on 6x and 6c.

TOTON95, 2023-10-30:
It seems to be fixed now, so I will close this issue.
[image]

closed 2023-10-30 by TOTON95