PX4/PX4-Autopilot #22265: SITL multi-vehicle ROS simulation, ACTUATOR_CONTROL_TARGET is not published

eng-86, 2023-10-26:
### Describe the bug

I am a PhD student. I am working on a distributive fault tolerant control for multi-UAVs. Each UAV can estimate its local actuator fault and neighbors fault relying on torques and thrust control inputs to each UAV in addition to UAVs' states. I was using a one-year-recent version PX4 with ROS to verify my proposed algorithms. It was working except that the real time scale in GAZEBO persists at 0.2 when trying to simulate four vehicles with an error "ERROR [param] Parameter XRCE_DDS_KEY not found.". This caused an accuracy problem (the algorithm was also validated through matlab simulation). When I upgraded the code to the latest one, the real time problem was solved, but a new problem appeared. My algorithm depends on desired mavros actuator target controls for each UAV to estimate the actuator fault, and it is not published in new versions (I checked it during simulation and according to the discussion also https://github.com/mavlink/mavlink/pull/1784 ). I compared also two log data samples. I found that in old version, the actuator control log was found (Roll, Pitch, Yaw, Thrust (up), Thrust (forward). 

### To Reproduce

cd PX4-Autopilot
git submodule update --init --recursive
DONT_RUN=1 make px4_sitl_default gazebo-classic
source Tools/simulation/gazebo-classic/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd):$(pwd)/Tools/simulation/gazebo-classic/sitl_gazebo-classic
roslaunch px4 multi_uav_mavros_sitl.launch

### Expected behavior

Publishing ACTUATOR_CONTROL_TARGET in both MAVLINK and MAVROS (Roll, Pitch, Yaw, Thrust (up), Thrust (forward). Those parameters as very important in research domain. I think it is important to publish such messages.

### Flight Log

[image]

### Software Version

1.14.0

### Flight controller

SITL

### Vehicle type

Multicopter

hamishwillee, 2023-11-01:
@sfuhrer FYI you removed this in #20837. @Jaeyoung-Lim feels this was suboptimal in https://github.com/mavlink/mavlink/pull/1784#issuecomment-1637936479. I don't personally have an opinion other than removing external interfaces should not be a surprise. Is this something that might come back?

eng-86, 2023-11-01:
> @sfuhrer FYI you removed this in #20837. @Jaeyoung-Lim feels this was suboptimal in [mavlink/mavlink#1784 (comment)](https://github.com/mavlink/mavlink/pull/1784#issuecomment-1637936479). I don't personally have an opinion other than removing external interfaces should not be a surprise. Is this something that might come back?

But as a researcher in academic domain, I think dealing with virtual controls is important while developing some algoritms. I think it is better to be added.

hamishwillee, 2023-11-01:
It might be that this was removed because there are better ways of doing this now. Let's see if @sfuhrer or others have advice.

eng-86, 2023-11-01:
> It might be that this was removed because there are better ways of doing this now. Let's see if @sfuhrer or others have advice.

I don't think that, I reviewed the code and Previous issues. Also I am not an expert, but I have been using px4 environment from about 5 years for developing actuator fault Tolerant algoritms, and I have published a paper based on it. Some algorithms and mainly observers depend on applied virtual control inputs.

eng-86, 2023-11-04:
@sfuhrer @hamishwillee , @Jaeyoung-Lim,  hello again, any new thoughts or ideas?

sfuhrer, 2023-11-09:
@eng-86 You're right, the functionality to set "roll", "pitch", "yaw" and "collective thrust" commands has been removed from PX4, as the previous way of doing it through MAVLink was no longer compatible with PX4. 
What we should do is add new MAVLink messages for 3D torque and 3D thrust.

eng-86, 2023-11-09:
@sfuhrer is there a plan for that? Is it easy for me to add it? If yes can you help please. Please note that I am working with mavros. I really need it to publish my paper. I validated my work through MATLAB, and since I am planning to publish in a good journal (IEEE transaction) they need some real validation.

eng-86, 2023-11-09:
@sfuhrer Additional information: I returned to PX4 13.3 release. I could see the target actuator control published, but an additional problem appeared: when I did some changes to the code and built it, the changes did not reflect on the results (in other words, no effect of changes). The changes are injecting actuator fault after a time from takeoff to validate my proposed detection algorithm. This is not the first time I perform such changes.

sfuhrer, 2023-11-10:
> @sfuhrer is there a plan for that? Is it easy for me to add it? If yes can you help please. Please note that I am working with mavros. I really need it to publish my paper. I validated my work [...]

> What we should do is add new MAVLink messages for 3D torque and 3D thrust.

That's the plan. As it's a MAVLink change I expect it to take a couple of weeks. For you I recommend you to use 1.13 if you're time constrained.

sfuhrer, 2023-11-10:
> @sfuhrer Additional information: I returned to PX4 13.3 release. I could see the target actuator control published, but an additional problem appeared: when I did some changes to the code and built [...]

I don't know what's going on, but it's not related to the issue about the missing mavlink message that you raised here right? I would then ask on Discord for help.

eng-86, 2023-11-10:
@sfuhrer Thank you, any idea or guidance about the problem of changes are not reflecting after building the PX4 firmware?

sfuhrer, 2023-11-13:
> @sfuhrer Thank you, any idea or guidance about the problem of changes are not reflecting after building the PX4 firmware?

No, no idea. But also that one I would not discuss here but on discord or a new github issue.

sfuhrer, 2023-11-13:
MAVLink proposal: https://github.com/mavlink/mavlink/pull/2055

hamishwillee, 2023-12-20:
@eng-86 There are some questions in https://github.com/mavlink/mavlink/pull/2055 that really need your help/context in order to progress, such as:
- https://github.com/mavlink/mavlink/pull/2055#discussion_r1428878636
- https://github.com/mavlink/mavlink/pull/2055#discussion_r1430556140
- https://github.com/mavlink/mavlink/pull/2055#discussion_r1428880428

Can you please advise there on your intent/needs?

eng-86, 2023-12-21:
> @eng-86 There are some questions in [mavlink/mavlink#2055](https://github.com/mavlink/mavlink/pull/2055) that really need your help/context in order to progress, such as:
> 
> * [development: add setter/getters for torque and thrust setpoints mavlink/mavlink#2055 (comment)](https://github.com/mavlink/mavlink/pull/2055#discussion_r1428878636)
> * [development: add setter/getters for torque and thrust setpoints mavlink/mavlink#2055 (comment)](https://github.com/mavlink/mavlink/pull/2055#discussion_r1430556140)
> * [development: add setter/getters for torque and thrust setpoints mavlink/mavlink#2055 (comment)](https://github.com/mavlink/mavlink/pull/2055#discussion_r1428880428)
> 
> Can you please advise there on your intent/needs?

Hello, sorry for late reply. I replied to the requests according to my experience. I think involving a contributor in control system will strengthen the answers.