PX4/PX4-Autopilot #22267: 8 Channel DShot (rather than 6)

kristanhilby, 2023-10-27:
### Describe problem solved by the proposed feature

I currently have 7 brushless motors that I want to run with DShot. I was able to get 6 of them up and running, but when I go to configure the 7th, it appears AUX channels 7 and 8 only allow PWM and OneShot (see attached image). Does anyone know how to configure to allow for DShot on all 8 AUX channels? Is there a limitation in the flight controller that would keep these additional channels from running DShot (I am using a Pixhawk 6C)? Thanks!

[image]

### Describe your preferred solution

Allow all 8 AUX channels (rather than 6) to command DShot!

### Describe possible alternatives

DShot on MAIN channels.

simopanzer89, 2024-04-26:
Hello,

@kristanhilby were you able to find a solution for this? We are exactly in the same situation.

I have been doing some research, and it looks like Ardupilot provides support for Dshot on Pixhawk 6C on all the 8 FMU channels based on their doc: https://ardupilot.org/copter/docs/common-holybro-pixhawk6C.html#pwm-output

@bkueng I found this related PR where you introduced support to Dshot for FMU-v5X: https://github.com/PX4/PX4-Autopilot/pull/13628

Do you think it is feasible to do the same for FMU-v6C? Is there any potential hardware conflict, or is the issue related to software?

Related information on PX4 discuss:

https://discuss.px4.io/t/pixhawk6c-dshot-capability-with-px4/31821
https://discuss.px4.io/t/pixhawk-6x-maximum-dshot-outputs/28123
https://discuss.px4.io/t/esc-telemetry-feedback/15122/8

kristanhilby, 2024-04-26:
Hi! Unfortunately not. Thankfully my ESCs allowed for PWM control as well so I just moved to using that + calibrating over using DShot (unfortunately). Best of luck!