PX4/PX4-Autopilot #22269: Upgrade nuttx to 12.3

mtal, 2023-10-27:
### Describe problem solved by the proposed feature

The latest stable nuttx version is 12.3. And it supports cmake.
PX4 still based on 10.x.

### Describe your preferred solution

Upgrade nuttx to 12.3

### Describe possible alternatives

Upgrade nuttx to 12.3

dagar, 2023-10-27:
There's no specific date, but we're definitely planning to upgrade NuttX at some point. Then for actually switching to NuttX cmake I'll have to review and rewrite out current integration, which I'm hoping will ultimately be much cleaner (and faster).

Violin9906, 2025-12-24:
Any update on this topic? I'm trying to port px4 to K230, which is supported by nuttx 12+