PX4/PX4-Autopilot #22272: [Bug] USB connection unstable with QGC whenever a SD card is not present (uploading FW from master)

TOTON95, 2023-10-29:
### Describe the bug

I was checking how `master` (on commit `12b291b82f4dcc2a8b1c59b86fddeee0b675abe2`) was doing for these targets:

- mRo Control Zero H7
- mRo Control Zero OEM H7
- mRo Pixracer Pro

*All above were using an Ardupilot ChibiOS bootloader

It runs successfully and gets connected to QGC (`4.2.9`) if an SD card is present, if not, it keeps skipping every auto-connect attempt, and manual ones as well. 

Occasionally, it will "connect" but also will display a "Connection lost" banner and at the same time give access to the usual setup menu, every attempt to order a sensor calibration for example, is rejected and the AP's buzzer throws an "error tone".

At first, it may look like that the issue might be located in a group of boards (mRo's in this case) but I also had laying around a Holybro Pixhawk 6C (with a PX4 FMU V6C) and the undesired behavior repeats after uploading the FW and SD card is removed. 

After this, I returned to `v1.14.0` release for every board without any issues, and operated normally with and without an SD card. 

Can someone who has the listed HW verify if this is true for you as well? Also, these verifications will be highly appreciated if anyone has other HW and some spare time. 

Please, let me know if I am missing something to solve this issue. Thank you in advance. 

# 

### To Reproduce

1. Make sure to have an SD card inserted into an AP board and plug it via USB into a computer running QGC `4.2.9`

2. Upload `master` FW (from source/QGC) into any of the listed boards 
3. Wait until the board reboots and inspect if it works correctly
4. Unplug the AP board and plug it back without the SD card
5. Observe interactions with QGC 
6. Insert back the SD card and observe interactions with QGC

### Expected behavior

The AP board should connect to QGC through USB and report back the status, sensors, shell, etc...

### Flight Log

Let me know if required, although it would be hard to get logs w/o SD card, unless you suggest another way 

### Software Version

NOTE: All these MAVLink Shell outputs executed on `master` (`1.15.0`) were obtained with a SD Card inserted into the AP, and the latest stable release (`1.14.0`) without an SD card where indicated

- mRo Control Zero H7

w/ 1.15.0 and SD card

```
HW arch: MRO_CTRL_ZERO_H7
PX4 git-hash: 12b291b82f4dcc2a8b1c59b86fddeee0b675abe2
PX4 version: 1.15.0 40 (17760320)
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: c23b72dffeb0de0d1a836ab561eb9169c4a9e58e
Build datetime: Oct 25 2023 21:07:48
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000600000000343538303332510a0036003f
MCU: STM32H7[4|5]xxx, rev. V
```

w/ 1.14.0 and w/o SD card
```
HW arch: MRO_CTRL_ZERO_H7
PX4 git-hash: b8c541dd7277ed735139d7d1bfb829d61fbe29fb
PX4 version: Release 1.14.0 (17694975)
[... 233 lines cut ...]
INFO  [uavcan] Node ID 1, bitrate 1000000
ERROR [uavcan] CentralizedServer init: -1

ERROR [uavcan] UavcanServers init: -1
NuttShell (NSH) NuttX-11.0.0
nsh> WARN  [health_and_arming_checks] Preflight Fail: Missing FMU SD Card
WARN  [health_and_arming_checks] Preflight Fail: Missing FMU SD Card
INFO  [gps] u-blox firmware version: SPG 3.01
INFO  [gps] u-blox protocol version: 18.00
INFO  [gps] u-blox module: NEO-M8N-0
INFO  [gps] u-blox firmware version: SPG 3.01
INFO  [gps] u-blox protocol version: 18.00
INFO  [gps] u-blox module: NEO-M8N-0
WARN  [health_and_arming_checks] Preflight Fail: Missing FMU SD Card
WARN  [health_and_arming_checks] Preflight Fail: Missing FMU SD Card
WARN  [health_and_arming_checks] Preflight Fail: Missing FMU SD Card
```

### Flight controller

mRo Control Zero H7, mRo Control Zero OEM H7, mRo Pixracer Pro, Holybro Pixhawk 6C

### Vehicle type

None

mrpollo, 2023-10-30:
Hey @dagar would you d taking as look?

mooey5775, 2024-02-28:
This seems like it was fixed with #22778

TOTON95, 2024-03-05:
@mooey5775 Thanks for noticing it. I will take a look in the upcoming days 🙌🏼

TOTON95, 2024-03-26:
@mooey5775 Can confirm that #22778 might fixed this issue for the following boards:
- Control Zero Classic
- Control Zero H7 OEM

I have a few more to test it. An update should be possible soon.

TOTON95, 2024-03-27:
I can add the PixRacer PRO to the list

TOTON95, 2024-03-27:
Finally, I can confirm that the fix works for Control Zero H7. Thank you @mooey5775

closed 2024-03-27 by TOTON95

mrpollo, 2024-03-27:
Thanks @TOTON95 🙌