PX4/PX4-Autopilot #22274: [Bug] Compilation error

zyy37, 2023-10-30:
### Describe the bug

Compilation error.
```sh
cd PX4-Autopilot
mkdir build1
cd build1
cmake -DCONFIG=px4_sitl_default -DCMAKE_BUILD_TYPE=Debug ..
make
```

The error logs are as follows,
[image]

In general, for most internet companies nowadays, they use cmake or ninja instead of Makefile. Please maintain the cmake way.

### To Reproduce

as above

### Expected behavior

Should compile successfully

### Flight Log

n/a

### Software Version

commit 794d0d177bf526a83b94c23f36b674f961955200

### Flight controller

px4_sitl_default

### Vehicle type

None

frede791, 2023-10-30:
Consider following the official PX4 guidelines to compile. Also note that PX4 does use CMake files. 
https://docs.px4.io/main/en/dev_setup/building_px4.html

zyy37, 2023-10-31:
> Consider following the official PX4 guidelines to compile. Also note that PX4 does use CMake files. https://docs.px4.io/main/en/dev_setup/building_px4.html

Thank you for your suggestion. I have referred to the document. It is mainly hoping to maintain the CMake way.

The error log seems to be that `build1/src/modules/uxrce_dds_client/dds_topics.h` was not found during build, and it should be that cmake did not generate it. Because I could build px4_sitl successfully by the cmake way ago. Did you reproduce it?

[image]

zyy37, 2023-10-31:
To avoid overwriting the original `build/` folder, I used cmake to build in another newly created folder `build1/`. Build file names are arbitrary, and that should be fine.

I mainly want to use cmake and `-DCMAKE_BUILD_TYPE=Debug` such two functions.

frede791, 2023-10-31:
I am not sure whether `Debug` is a valid string for `DCMAKE_BUILD_TYPE`. Have you consolidated the Makefile?
https://github.com/PX4/PX4-Autopilot/blob/main/Makefile

zyy37, 2023-10-31:
[image]
[image]

cmake has `Debug` attribute.

zyy37, 2023-11-02:
The `build1/src/modules/uxrce_dds_client/dds_topics.h` file was not generated because the `build1/src/modules/uxrce_dds_client/CMakeFiles/modules__uxrce_dds_client.dir/build.make` file was not executed. It generated the `dds_topics.h` file only after I executed the command manually.
[image]

Please check it, thanks. @frede791 , @beniaminopozzan, @mrpollo