PX4/PX4-Autopilot #22275: [Bug] Wrong rotor and faulty mesh

frede791, 2023-10-30:
### Describe the bug

In the rc_cessna model, the turning direction of the rotor is set to be clockwise (`cw`), however the .dae mesh model that is used is `iris_prop_ccw.dae` which is incorrect. 

While using the `iris_prop_cw.dae` would solve the issue, this mesh in itself has some flaws at there are pieces of the rotor missing when viewed from the back (but not the front). 

While this does not impact flight performance, it is an ugly visual bug. 

### To Reproduce

Change the 'rotor_puller_visual' mesh uri from `iris_prop_ccw.dae` to `iris_prop_cw.dae` and launch gazebo as normal with: `make px4_sitl gz_rc_cessna`.

### Expected behavior

Full rotor should be visible from front and back.

### Screenshot / Media

[image]

### Flight Log

N/A

### Software Version

Current main branch, commit: 794d0d177bf526a83b94c23f36b674f961955200

### Flight controller

It's in simulation.

### Vehicle type

None

Jaeyoung-Lim, 2023-10-30:
@frede791 We could also just reverse the turning direction of the rotor since it only has one rotor right?

frede791, 2023-10-30:
@Jaeyoung-Lim  Yes this would be a better solution, as the rotors on a Cessna are supposed to spin in a clockwise direction when viewed from the rear. Gazebo defines the rotation when viewed from the front.

closed 2023-11-09 by Jaeyoung-Lim