PX4/PX4-Autopilot #22283: Why Step Response for (Roll/Pitch/Yaw) Rate is Zero?

MuhammadBilal1, 2023-10-31:
### Describe the bug

Hi,
    I am using PX4 ([1.14.0](https://github.com/PX4/PX4-Autopilot/releases/tag/v1.14.0)) in my drone which is using [Pixhawk 6C](https://holybro.com/products/pixhawk-6c), but since last few days, I am getting an issue:
The Step Response for (Roll/Pitch/Yaw) Rate is **Zero** in the flight logs (PID Analysis).
I need this doing the PID Tuning of my drone.

**1-** Sometimes "Step Response for Roll Rate" at ZERO line
**2-** Sometimes "Step Response for Pitch Rate" is at ZERO line
**3-** Sometimes "Step Response for Yaw Rate" is at ZERO line
**4-** Sometimes "Step Response for Roll/Pitch Rate" is at ZERO line

Why is this happening? I am doing manual PID Tuning, flight behavior seems fine (not 100%, but not shaking or crashing), but response is zero.
I am attaching some of the logs here:

### To Reproduce

Performing manual tuning.
Almost all of the flights are done in **Position** flight mode.
I am using **UM982** dual antenna GPS.

### Expected behavior

The PID Analysis graphs should show the correct values of the PID for each rate instead of a flat line at 0.

### Screenshot / Media

[image]

[image]

### Flight Log

**Roll & Pitch** are at 0
https://review.px4.io/plot_app?plots=pid_analysis&log=60815230-a1fb-40fd-a64b-54abfcb1d143

**Roll, Pitch & Yaw** at 0
https://review.px4.io/plot_app?plots=pid_analysis&log=08c14c20-38ca-404e-aee4-30c531839f56

**Pitch** at 0
https://review.px4.io/plot_app?plots=pid_analysis&log=f804b7d3-5bf6-49cc-bef6-3473c7e4d8b4

**Roll & Pitch** are at 0
https://review.px4.io/plot_app?plots=pid_analysis&log=edcf0b09-5ea0-4dce-a5d5-115349aea314

**Roll, Pitch & Yaw** are available
https://review.px4.io/plot_app?plots=pid_analysis&log=4e796e69-30c9-4199-9063-81c6259ba9f6

### Software Version

PX4 ([1.14.0](https://github.com/PX4/PX4-Autopilot/releases/tag/v1.14.0))

### Flight controller

Pixhawk 6C

### Vehicle type

Multicopter

MuhammadBilal1, 2023-11-01:
@bresch Can you please help me here!

bresch, 2023-11-01:
@MuhammadBilal1 This is most likely a https://github.com/PX4/flight_review issue.

MuhammadBilal1, 2023-11-01:
> @MuhammadBilal1 This is most likely a https://github.com/PX4/flight_review issue.

So shall I create an issue on the **Flight Review** repository?

bresch, 2023-11-02:
> So shall I create an issue on the Flight Review repository?

Yes, I don't think we can really help you here.

closed 2023-11-15 by MuhammadBilal1