PX4/PX4-Autopilot #22285: [Bug] rc_cessna cannot be launched without having a ground station running.

frede791, 2023-10-31:
### Describe the bug

The rc_cessna model cannot take off without also having a ground station open. All other models present in PX4 for Gazebo do not require a ground station to be open and can be commandeered solely from the PX4 command line. 

### To Reproduce

Run `make px4_sitl gz_rc_cessna` while not running QGC or AMC. 

### Expected behavior

Model should be able to arm and provide the `Ready for takeoff!` prompt after which the vehicle can be launched through the PX4 command line. 

### Screenshot / Media

[image]

### Flight Log

N/A

### Software Version

Current PX4 main branch, Commit: 26fd4c852c52837e306790e0f86b27d0b0df516e

### Flight controller

It's in simulation.

### Vehicle type

Fixed Wing

dagar, 2023-10-31:
In the past my opinion was that we should keep the simulations as closely in sync with real usage as possible (eg data link failsafes) to prevent any surprises when people make the transition. The downside of this is the annoyance of requiring QGC for a quick test.

What do you think?

Jaeyoung-Lim, 2023-10-31:
@frede791 This is realistic behavior of the vehicle, and a mere inconvenience for simulation development.

IMHO, SITL only makes sense if it replicates the behavior of real vehicles.

dagar, 2023-10-31:
> I would aim to enforce consistency across all models. So either all of them don't run without QGC or they all do.

Yes definitely, anything that's not I think we should consider a bug.

frede791, 2023-10-31:
@dagar @Jaeyoung-Lim So which one do you wanna go for?

Jaeyoung-Lim, 2023-10-31:
We should keep the current behavior

dagar, 2023-10-31:
I agree with @Jaeyoung-Lim.

Somewhat contradicting myself here, but I'm not opposed to having some kind of alternative for this use case.
 - maybe we could have an alternate airframe explicitly called x500_no_gcs? 
 - find a way to cheat like having the console become a stand-in for manual control?
    - here's a somewhat silly idea I had for CLI manual_control https://github.com/PX4/PX4-Autopilot/pull/17681, but we could easily add a few more things like this aerostack2 keyboard teleop https://aerostack2.github.io/_06_user_interfaces/_keyboard_teleoperation/index.html#introduction

frede791, 2023-10-31:
I don't know about adding more models that add nothing in terms of behavior or appearance. I feel that this would be better solved through setting a env variable somewhere that tells px4 it is allowed to run without ground control.

danielhonies, 2023-11-06:
> In the past my opinion was that we should keep the simulations as closely in sync with real usage as possible (eg data link failsafes) to prevent any surprises when people make the transition. The downside of this is the annoyance of requiring QGC for a quick test.
> 
> What do you think?

Regarding consistency, the parameter is not set for the real world airframes though.

Jaeyoung-Lim, 2023-11-06:
@danielhonies Where is this being set? We should change that

danielhonies, 2023-11-06:
@Jaeyoung-Lim It is set in the airframe files as a default param. Check the PR to see where he changed it.
Should we really set this though? I feel like a decent amount of people still fly real vehicles with a conventional RC connection and no GCS, but from what I read from the code this would then end up in a pre-flight fail.

Jaeyoung-Lim, 2023-11-06:
> Should we really set this though? I feel like a decent amount of people still fly real vehicles with a conventional RC connection and no GCS, but from what I read from the code this would then end up in a pre-flight fail.

At least for me, enabling it would be the safer option. I wouldn't assume someone flying with no GCS connection without any feedback would know what they are doing - unless they opt-in to this option. They can still fly without a GCS connection. They just can't do it without understanding that PX4 has different flight modes, and the behavior is not fully defined with the four stick inputs.

danielhonies, 2023-11-06:
You can see it here: https://github.com/PX4/PX4-Autopilot/pull/22299/files, it's the NAV_DLL_ACT param. A GitHub search through the whole repo only brings up a few simulation airframes where it is set.

danielhonies, 2023-11-06:
In my use case we would just configure everything with GQC and map the flight modes to an RC Channel. I'm thinking Hobbyists/FPV here. Yeah no real feedback is not optimal but I think a lot of beginners would also not know why their drone is not arming and be really confused.

frede791, 2023-11-08:
I agree with @danielhonies that setting the requirement for real world airframes would only lead to confusion for beginners. I would therefore opt to only enforce it across all simulation airframes.

PR #22299 (2024-06-11): update GCS connection loss failsafe in all gazebo models

### Solved Problem
When launching the rc_cessna in Gazebo I found that the model would not fly unless a ground station was also connected. This was not the case for any of the other models. 

Fixes #22285 

### Solution
- After discussion in the issue thread, it was decided that the current behaviour of the cessna should be kept. This PR applies the cessna behavior to all other gazebo simulation vehicles in px4. 

### Changelog Entry
For release notes: Launching a vehicle in Gazebo requires a connection to a ground station. 
```

### Al [... cut]
closed 2024-06-11 by sfuhrer