PX4/PX4-Autopilot #22292: [Bug] Raw data of external compass

fbenti, 2023-11-01:
### Describe the bug

I connect an external RM3100 compass to a px4 board and calibrated it. 

If I check the data from the MAVLINK console, everything seems fine: I can see two different compass data.
[image]

What I would like to have is to have this raw data transported on the ros2 side through the xrce agent. Therefore, I modified the `dds_topic.yaml` adding the following to the publishers' list.
```
  - topic: /fmu/out/sensor_mag
    type: px4_msgs::msg::SensorMag
```

I'm then actually able to see the topic, however only the data relative to the internal compass are published, while I would like to get the external one. 
[image]

Is this behavior expected? How can I change such that both are published?
connected

### To Reproduce

-

### Expected behavior

-

### Flight Log

 -

### Software Version

Stable release: v1.14.0

### Flight controller

PX4 FMUv5

### Vehicle type

None

junwoo091400, 2023-11-06:
Not sure if multi topics are supported for ROS2 DDS topics side, @beniaminopozzan do you have comment on this?

beniaminopozzan, 2023-11-13:
Hi @fbenti , that feature is not available in `v1.14`, but you can try https://github.com/PX4/PX4-Autopilot/pull/22350 :)

closed 2025-11-18 by beniaminopozzan