PX4/PX4-Autopilot #22296: [Bug] System Failure Injection (gps) not working

hamishwillee, 2023-11-02:
### Describe the bug

I'm trying to inject GPS failure in simulator following instructions here: https://docs.px4.io/main/en/simulation/failsafes.html#sensor-system-failure

However I get a failure on the command - getting ACK. On QGC I still appear to have gps.

[image]

### To Reproduce

WSL2 simulation of current main on Windows, connected to QGC. 

In PX4 shell enter:
```
param set SYS_FAILURE_EN 1
failure gps off
```
Get the error that no ack is received. QGC still shows valid GPS and position. I can still take off and arm.

### Expected behavior

I'd expect the `failure gps off` to trigger GPS fail visible both in PX4 shell and QGC.

### Screenshot / Media

[image]

### Flight Log

https://logs.px4.io/plot_app?log=f2ed9706-dee8-40af-ab78-7d7fa02c0abe

### Software Version

Main running in WSL2

### Flight controller

NA

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

NA

### Additional context

@julianoes You might have advice on this?

julianoes, 2023-11-02:
Which simulation is that? My assumption is that it's not implemented for that one.

hamishwillee, 2023-11-03:
Probably Gazebo (gz). We're going to need everything in that simulator eventually.

julianoes, 2023-11-04:
Right, so it's implemented for classic here:
https://github.com/PX4/PX4-Autopilot/blob/22ee90b7d7744012e54abacb1679f4c06b899ad6/src/modules/simulation/simulator_mavlink/SimulatorMavlink.cpp#L1232-L1235

But presumably not gz (that new thing people talk about :smile: )

closed 2026-08-28 by dakejahl