PX4/PX4-Autopilot #22300: [Bug] Running make and binary simultaneously leads to error

frede791, 2023-11-02:
### Describe the bug

When running `PX4_GZ_MODEL=gz_x500 make px4_sitl gz_x500`, Gazebo opens but the model cannot be found and the simulation crashes. 
`[Err] [UserCommands.cc:1097] Error Code 1: Msg: Error finding file [gz_x500/model.sdf].` is thrown. 

### To Reproduce

Run `PX4_GZ_MODEL=gz_x500 make px4_sitl gz_x500`
This can also be done with any other model or two dfferent models. 

### Expected behavior

Setting the env variable: `PX4_GZ_MODEL` should not invalidate the model unless the env variable model is different from the model specified in the make command. Even when the two are different, one should take precedence over the other and Gazebo should not crash. 

### Flight Log

N/A

### Software Version

Current PX4 main

### Flight controller

It's in simulation.

### Vehicle type

None

beniaminopozzan, 2023-11-02:
Hi @frede791 
The `gz_` prefix is used with `make px4_sitl gz_x500` to distinguish between the different simulators targets.
But the actual model name is just `x500`. If you set `PX4_GZ_MODEL=gz_x500` then Gazebo looks for a model named `gz_x500`, which does not exists.
In conclusion, you should use
```
PX4_GZ_MODEL=x500 make px4_sitl gz_x500`
```

A few references:

https://github.com/PX4/PX4-Autopilot/blob/d6dbf38a1ba376cde9e65c10430e6feb52f6fdd5/src/modules/simulation/gz_bridge/CMakeLists.txt#L145-L146
is where the `gz_x500` target is defined. You can see that it sets `PX4_SIM_MODEL=gz_x500`

https://github.com/PX4/PX4-Autopilot/blob/d6dbf38a1ba376cde9e65c10430e6feb52f6fdd5/ROMFS/px4fmu_common/init.d-posix/rcS#L37-L65
Here the startup scripts looks for the `SYS_AUTOSTART` which is needed to choose the appropriate airframe.
`PX4_SIM_MODEL` is then used to infer `SYS_AUTOSTART` (unless other conditions hold)
https://github.com/PX4/PX4-Autopilot/blob/d6dbf38a1ba376cde9e65c10430e6feb52f6fdd5/ROMFS/px4fmu_common/init.d-posix/rcS#L56-L57

If `PX4_GZ_MODEL` is defined, the startup scripts uses it to spawn the model (and the world, possibly).
https://github.com/PX4/PX4-Autopilot/blob/d6dbf38a1ba376cde9e65c10430e6feb52f6fdd5/ROMFS/px4fmu_common/init.d-posix/px4-rc.simulator#L109-L110
If it is not defines, then `PX4_SIM_MODEL` is used, but first the `gz_` suffix is removed!
https://github.com/PX4/PX4-Autopilot/blob/d6dbf38a1ba376cde9e65c10430e6feb52f6fdd5/ROMFS/px4fmu_common/init.d-posix/px4-rc.simulator#L162-L163

frede791, 2023-11-02:
Ok thanks that makes sense. Thanks

closed 2023-11-02 by frede791