PX4/PX4-Autopilot #22306: [Bug] Version 14 - Parameter Modification Cause Reboot

YuxiaHu, 2023-11-03:
### Describe the bug

A drone (coaxial rotor) with PX4 firmware version 14 is connected to QGC. When parameters are modified on QGC, the drone will reboot from time to time and then disconnect. For example, if three parameters are modified, sometimes the first two are modified successfully, but the last one is not, and sometimes none of them is modified successfully. The situation in which the drone reboots due to parameter modification is not fixed.

But this has never happened in version 12. When it was changed to version 14, the phenomenon happened (both on QGC v4.1.2 and v4.2.8, both on board fmu-v5 and cubepilot cubeorange).

### To Reproduce

/

### Expected behavior

/

### Flight Log

/

### Software Version

PX4: release 1.12 and release 1.14
QGC: v4.1.2 and v4.2.8

### Flight controller

fmu-v5 and cubepilot cubeorange

### Vehicle type

Other

danielhonies, 2023-11-06:
Is the flight controller hard faulting?

YuxiaHu, 2023-11-08:
In fact, we tried two boards, v5 is new and should have no hard faulting.

closed 2024-01-16 by YuxiaHu