PX4/PX4-Autopilot #22308: [Bug] Rover cannot move by RC controller to stick input on the main screen of QGroundControl

giangthewalkingman, 2023-11-04:
### Describe the bug

I install the 1.14 stable version of PX4 into my Pixhawk 6x model  on  my rover. The parameters are ok,  however, I cannot control my  car via my RC or the sticks on the main screen of QGroundControl. When I use an oscilloscope to check, the PWM out stays at 1500 at both  MAIN1 and MAIN2 instead of adjusting its value while RC value keeps adjusting. 
In contrast, at 1.13.3 firmware version, the rover acts really okay.
[bug_04_11_23.txt](https://github.com/PX4/PX4-Autopilot/files/13256479/bug_04_11_23.txt)
Additionally, QGroundControl says the parameter CP_DIST is missed because this firmware is not well supported.

### To Reproduce

 at 1.13.3 firmware version, the rover acts really okay.

### Expected behavior

the rover acts normally like throttling forward when I move the stick of RC channel 1 forward.

### Flight Log

[log_40_2023-11-4-04-12-28.zip](https://github.com/PX4/PX4-Autopilot/files/13256508/log_40_2023-11-4-04-12-28.zip)

### Software Version

1.14.

### Flight controller

I use TaranisQX7 Flight controller and X8R receiver. I don't think they matter because my rover acts normally at earlier version like 1.13.3.

### Vehicle type

Rover

### How are the different components wired up (including port information)

MAIN1 -  ESC1
MAIN2 -  ESC2

eogks8921, 2023-11-22:
When building version 1.14.0 rc-1 using pixhawk6X, the rover module is missing. Adding it will make it work. #22130

AlbertoJimenezUC3M, 2024-02-28:
Hello, has anyone been able to solve it? I can´t use a rover with 1.14 :(

AlbertoJimenezUC3M, 2024-03-07:
If anybody has the same problem, I could solve it with the help of @eogks8921.
To solve this problem: you need to write CONFIG_MODULES_ROVER_POS_CONTROL=y in default.px4board (of your specific board) and build again. If you face some kind of memory error, delete no use modules.