PX4/PX4-Autopilot #22312: Combine follow me with the obstacle avoidance

brunopinto900, 2023-11-05:
### Describe problem solved by the proposed feature

Currently PX4 offers the Follow me flight mode and obstacle avoidance. Is there any plans to combine both? Similar to Skydio (of course within PX4 hardware limitations).

### Describe your preferred solution

I would love to contribute to this.

### Describe possible alternatives

-

junwoo091400, 2023-11-06:
This makes a lot of sense. But cases differ depending on what you mean by obstacle avoidance:

1. [Obstacle avoidance](https://docs.px4.io/main/en/computer_vision/obstacle_avoidance.html) (as companion computer): It currently works only with Offboard & Mission mode
2. [Collision Prevention](https://docs.px4.io/main/en/computer_vision/collision_prevention.html): It currently works only with Position mode

So overall, PX4 would need to adapt to enable obstacle avoidance / collision prevention even in follow me mode. Since collision prevention would be an easier case to modify, I think the solution could be following:

1. Create higher level collision prevention setpoint modifier that can incorporate collision prevention feature to all autonomous flight modes (Position, Follow me, Mission, etc) for a Multirotor
2. Apply that modifier logic as part of the flight mode manager to always have it in effect if possible.

@MaEtUgR thoughts on this?

p.s. I investigated and found that [MPC_POS_MODE](https://docs.px4.io/main/en/computer_vision/collision_prevention.html#px4-software-setup) needs to be set to either 0 (Manual Velocity) or 3 (Manual jerk limited smoothed Velocity), and not 4 (Manual Acceleration). But to me it seems like all position modes eventually generate the velocity setpoint, so I don't know why we have onlly allowed CP for two out of 3 Position control modes.

Related: https://github.com/PX4/PX4-Autopilot/issues/17240

dirksavage88, 2023-11-07:
@junwoo091400 I recently saw a video from the summit where Ronal George talked about a motion planner using the matlab toolbox for UAV's. He mentioned a [vector field histogram ](https://www.mathworks.com/help/nav/ug/vector-field-histograms.html)

Perhaps a local planner could exist as a module in PX4 and pull in the obstacle_distance sensor messages. I am not sure how compute intensive these algorithms are. 

See arducopter's bendy rule & djikstra's algorithms[ here](https://ardupilot.org/copter/docs/common-oa-dijkstrabendyruler.html)

OgnjenX, 2024-01-24:
I'm interested in the local obstacle avoidance feature. So I see you are mentioning possible approaches for accomplishing something like that @dirksavage88 
My current thoughts on that are these. I currently see two options for implementing local obstacle avoidance (without a companion computer):

1. Collision prevention already uses the 'Vector Field Histogram' algorithm for limiting velocity setpoints based on requested velocity and distance of near obstacles. I'm not sure if this approach is feasible, but I thought about reusing that existing functionality for obstacle avoidance (having a single instance that can be used and updated in multiple places - collision prevention and local obstacle avoidance algorithms)

2. Creating a new module that will be a new px4/nuttx task (likely with the lowest priority) that would receive the requested trajectory and update it based on near obstacles and publish the updated trajectory back - which is used by the ObstacleAvoidance class (very similar to how it currently works, but only with companion computer)

What are the opinions of other folks on that topic? Is anyone else interested in local obstacle avoidance?