PX4/PX4-Autopilot #22318: [Bug] position mode switches to hold and does not allow to move horizontally

zeroos, 2023-11-06:
### Describe the bug

I took of the drone in position mode. It is possible to change the altitude and yaw, but horizontal movements don't work.

### To Reproduce

Drone switched on and set to position mode.
Take off.
Try to move forward.
The drone won't move.

### Expected behavior

The drone should move forward.

### Flight Log

https://logs.px4.io/plot_app?log=e2979d85-f253-47df-b0a2-50855762ef59

### Software Version

1.15alpha (flashed using the "developer build" option from QGC)

### Flight controller

Pixhawk 6C, PX4 Vision drone

### Vehicle type

Multicopter

zeroos, 2023-11-07:
We've found out what was wrong: by default, PX4 Vision does not work without the obstacle avoidance software running on the companion board, and by mistake we reset some parameters to default.

For anyone in the future having the same problem with this or another drone, the parameters we needed to change were:

    COM_OBS_AVOID = False
    CP_DIST = -1

closed 2023-11-07 by zeroos

MaEtUgR, 2023-11-07:
Oh I see. I already had the log open. Some part of the flight controller (probably the obstacle avoidance part then) tried to continuously put the drone into hold mode. But the stick was always moved and switched it back to position. Following your findings I would further guess that collision prevention prevented you from moving because unkown surouning equals obstacle :man_shrugging: 
[image]
I'm glad you found the problem :+1: