PX4/PX4-Autopilot #22323: [Bug] uxrce_dds_client not receiving any packets and FILLED buffer gets > 93%

kyuhyong, 2023-11-07:
### Describe the bug

Running px4 release/1.14 on 6X module.
Offboard PC(OPC) is running ubuntu 18.04 with ROS2 foxy installed. 
Host PC(HPC) is running ubuntu 20.04 with ROS2 foxy apt installed.

PX4 <-- serial --> OPC <-- ethernet --> HPC

When both micro_ros_agent and uxrce_dds_client are running,
uxrce client shows connected but no rx packet received. 
```
nsh> uxrce_dds_client status
INFO  [uxrce_dds_client] Running, connected
INFO  [uxrce_dds_client] Using transport: serial
INFO  [uxrce_dds_client] Payload tx: 52371 B/s
INFO  [uxrce_dds_client] Payload rx: 0 B/s
```
Task shows 93.3% FILLED in nsh
```
 600   600 100 FIFO     Task    --- Waiting  Signal    00000000 009880 009220  93.3%! uxrce_dds_client start -t serial -d /dev/ttyS3 -b 921600
```
I tried to send arming command but nothing changed.

### To Reproduce

On PX4, start uxrce_dds_client
```
uxrce_dds_client start -t serial -d /dev/ttyS3 -b 921600
```
On OPC, start micro_ros_agent
```
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyHS1 -b 921600
```
On HPC, start velocity commander
```
ros2 launch px4_offboard offboard_velocity_control.launch.py
```

### Expected behavior

PX4 would arm when commanded to arm the vehicle via ros 2 topic.
Ros2 Topic echo shows command has sent to the agent as below
```
$ ros2 topic echo /fmu/in/vehicle_command
timestamp: 1699321903761104
param1: 1.0
param2: 0.0
param3: 0.0
param4: 0.0
param5: 0.0
param6: 0.0
param7: 0.0
command: 400
target_system: 1
target_component: 1
source_system: 1
source_component: 1
confirmation: 0
from_external: true
---
```

### Screenshot / Media

[image]

### Flight Log

No log is available

### Software Version

release/1.14

### Flight controller

6X

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

px4 release/1.14 on 6X module.
Offboard PC(OPC) is running ubuntu 18.04 with ROS2 foxy installed. 
Host PC(HPC) is running ubuntu 20.04 with ROS2 foxy apt installed.

PX4 <--(dev/ttyS3) serial (dev/ttyHS1)--> OPC <-- wired ethernet --> HPC

beniaminopozzan, 2023-11-22:
Hi @kyuhyong , just for confirmation, are you workink on the branch `release/1.14`?

kyuhyong, 2023-11-23:
> Hi @kyuhyong , just for confirmation, are you workink on the branch `release/1.14`?

yes that's correct. As mentioned in https://discuss.px4.io/t/uxrce-dds-client-filled-up-to-93-1/35177/4 
This issue was gone after stop manually starting the client from nsh but start by PARAM setting.

Uzingr, 2024-06-11:
Hi @kyuhyong i have a similar error, the micro xrce_dds client don't receive anything from xrce agent. In particular when i send message from topic i receive this error:
[image]
xrce_dds_client: Pixhawk 6C, xrce_dds_agent: a companion Computer. I launch a ROS file from a Ground computer.
You have any idea?

kyuhyong, 2024-06-12:
Hi @Uzingr  Did you check xrce_dds_agent is up and running on nsh? 
Also need to check xrce dds is installed both on CC and GC.

Uzingr, 2024-06-14:
Hi @kyuhyong i changed the baud rate to 921600 but didn't work. I think is a problem of this repo: https://github.com/ARK-Electronics/ROS2_PX4_Offboard_Example/blob/master/README.md because if i use a simple offboard_control example (https://docs.px4.io/main/en/ros2/offboard_control) by PX4_ros_com the client receives something (like 750 B/s).
It's more strange because the repo that i use worked in SITL.
Another problem is when i use the offboard_control example and launch the XRCE-DDS agent the drone give me this error: "no offboard signal". Why i receive this error?

albertocastrovill, 2024-10-08:
Hi @Uzingr, I am running the XRCE-Agent on my CC (raspberry pi) and using 921600 baudrate. When I type "uxrce_dds_client status" it returns me "running, connected, Payload TX: 48458 , Payload RX:0. And after running "ros2 topic list" no topic /fmu/in/ or /fmu/out/ appears
 Did you find out how to fix it?

tuananh0606-tech, 2025-02-26:
@albertocastrovill Did you find out how to fix it?

mrpollo, 2025-02-27:
You shouldn't post on other peoples issues, it's not going to help you. Please create a new issue with as much details as possible.

closed 2025-02-27 by mrpollo

rosiakpiotr, 2025-08-31:
I see this issue was closed as "not planned" but if anybody stumbles upon that problem verify that you have domain ID set to the same value on both, autopilot and ros2 cause that ate solid day of debugging in my case! I use Docker setup and posted the service with agent from microros dockerhub registry here:

https://discuss.px4.io/t/microxrceagent-created-px4-topics-not-shown-in-ros2-topic-list/45194/3