PX4/PX4-Autopilot #22337: [Bug] Advanced Lift Drag module not found

frede791, 2023-11-09:
### Describe the bug

When trying to run the advanced plane, the following error will occur:

```
Err] [SystemLoader.cc:119] Failed to load system plugin: (Reason: Could not find shared library)
- Requested plugin name: [gz::sim::systems::AdvancedLiftDrag]
- Requested library name: [gz-sim-advanced-lift-drag-system]
- Library search paths:
  - /home/ubuntu/.gz/sim/plugins/
  - /usr/lib/x86_64-linux-gnu/gz-sim-7/plugins/
  - /home/ubuntu/.ignition/gazebo/plugins/
  ```
This is because Gazebo currently does not support the Advanced Lift Drag Plugin. Once it is part of the distribution, this error will disappear. 

### To Reproduce

`make px4_sitl gz_advanced_plane`

### Expected behavior

Advanced Plane should start.

### Flight Log

N/A

### Software Version

Current Main

### Flight controller

It's in simulation.

### Vehicle type

None

### Additional context

This issue does not have to be solved as there is currently nothing to be done. It is only here to track the issue until it is resolved.

frede791, 2023-12-07:
Update (7.11.2023): No new Gazebo release yet, so this is still active.

Jaeyoung-Lim, 2023-12-07:
@ahcorde Would it be possible to shed some light on the plan for a new release for gazebo?

closed 2025-03-01 by dakejahl