PX4/PX4-Autopilot #22351: [Bug] Build Error for Custom SITL Airframe in PX4: Unknown Target

alireza787b, 2023-11-13:
### Describe the bug

Encountering a build error when trying to add a custom SITL airframe in PX4. Despite following standard procedures for creating and integrating a new airframe script (22001_none_plane), the build system returns an "unknown target" error.

### To Reproduce

Created a script 22001_none_plane in ROMFS/px4fmu_common/init.d-posix/airframes, duplicating the standard plane script from init.d.
Added 22001_none_plane to CMakeLists.txt in init.d-posix/airframes.
Ran make distclean and make clean.
Executed the build command: make px4_sitl none_plane.
Received the error: ninja: error: unknown target 'none_plane'.

### Expected behavior

The build system should recognize the custom airframe and compile it without errors, as per the standard process for adding new SITL airframes. just like make px4_sitl none_iris

### Flight Log

there is no log

### Software Version

-- PX4 version: v1.15.0-alpha1-80-g1592aedc11 (1.15.0)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3") 
-- PX4 config file: /home/alireza/PX4-Autopilot/boards/px4/sitl/default.px4board
-- PLATFORM posix
-- ROMFSROOT px4fmu_common
-- ROOTFSDIR .
-- TESTING y
-- ETHERNET y
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- The CXX compiler identification is GNU 11.4.0
-- The C compiler identification is GNU 11.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- cmake build type: RelWithDebInfo
/usr/bin/python3: Error while finding module specification for 'symforce.symbolic' (ModuleNotFoundError: No module named 'symforce')
-- Could NOT find gz-transport (missing: gz-transport_DIR)
-- Looking for gz-transport12 -- found version 12.2.1
-- Searching for dependencies of gz-transport12
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so (found version "3.12.4") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.3.4
-- Found ZeroMQ: TRUE (Required is at least version "4") 
-- Checking for module 'uuid'
--   Found uuid, version 2.37.2
-- Found UUID: TRUE  
-- Looking for gz-utils2 -- found version 2.0.0
-- Searching for dependencies of gz-utils2
-- Searching for <gz-utils2> component [cli]
-- Looking for gz-utils2-cli -- found version 2.0.0
-- Searching for dependencies of gz-utils2-cli
-- Looking for gz-msgs9 -- found version 9.5.0
-- Searching for dependencies of gz-msgs9
-- Looking for gz-math7 -- found version 7.3.0
-- Searching for dependencies of gz-math7
-- Looking for gz-utils2 -- found version 2.0.0
[... 89 lines cut ...]
-- Configuring done
-- Generating done
-- Build files have been written to: /home/alireza/PX4-Autopilot/build/px4_sitl_default
ninja: error: unknown target 'none_plane'
make: *** [Makefile:232: px4_sitl] Error 1

### Flight controller

Not applicable (Software In The Loop simulation on WSL - Windows Subsystem for Linux).

### Vehicle type

Fixed Wing

### How are the different components wired up (including port information)

working in WSL2 in windows 11

### Additional context

Followed guidelines and examples for creating custom SITL airframes in PX4.
Searched for sitl_target.cmake in platforms/posix/cmake as per documentation but couldn't find it in the latest version.
This process worked for others in the community, suggesting a potential bug or a change in the build system in the latest PX4 version.
Running PX4 on WSL might have specific considerations; all file format checks (line endings, permissions) have been verified.
right now make px4_sitl none_iris works and I have connected it to my costum simulator. now I want to have a fixed wing instead of iris airframe.

alireza787b, 2023-11-25:
I managed to identify and solve the problem myself. It appears that there have been changes in the handling of airframe names in PX4, particularly from version 1.15 RC onwards. In older versions, one would typically define a 'plane' airframe and then use 'none_plane' for additional configurations. However, in the latest version, each simulator definition needs to be specified separately.

To address this, I had to define a 'none_plane' airframe. I noticed that there were some updates and missing steps in the documentation, specifically in modifying the `src/modules/simulation/simulator_mavlink/CMakeLists.txt` file. Here is the necessary addition:

```cmake
# none_plane (legacy compatibility launch helper)
add_custom_target(none_plane
	COMMAND ${CMAKE_COMMAND} -E env PX4_SYS_AUTOSTART=22001 $<TARGET_FILE:px4>
	WORKING_DIRECTORY ${SITL_WORKING_DIR}
	USES_TERMINAL
	DEPENDS
		px4
		${PX4_SOURCE_DIR}/ROMFS/px4fmu_common/init.d-posix/airframes/22001_none_plane
	COMMENT "launching px4 none_plane (SYS_AUTOSTART=22001)"
)
```

Additionally, it's important to include 'none_plane' in the `cmakelists` in the `init.d-posix/airframes/` directory as well.

For reference, my airframe definition is as follows:

```sh
#!/bin/sh
#
# @name Generic Standard Plane SITL for custom Simulator
#
# @type Standard Plane
# @class Plane
# @author alireza787b
#

. ${R}etc/init.d/rc.fw_defaults

param set-default CA_AIRFRAME 1
param set-default CA_ROTOR_COUNT 1
param set-default CA_ROTOR0_PX 0.3
param set-default CA_SV_CS_COUNT 4
param set-default CA_SV_CS0_TYPE 1
param set-default CA_SV_CS0_TRQ_R -0.5
param set-default CA_SV_CS1_TRQ_R 0.5
param set-default CA_SV_CS1_TYPE 2
param set-default CA_SV_CS2_TRQ_P 1.0
param set-default CA_SV_CS2_TYPE 3
param set-default CA_SV_CS3_TRQ_Y 1.0
param set-default CA_SV_CS3_TYPE 4

param set-default SENS_EN_GPSSIM 1
param set-default SENS_EN_BAROSIM 1
param set-default SENS_EN_MAGSIM 1
param set-default SENS_EN_ARSPDSIM 1

# disable some checks to allow to fly:
# - with usb
param set-default CBRK_USB_CHK 197848
# - without real battery
param set-default CBRK_SUPPLY_CHK 894281
# - without safety switch
param set-default CBRK_IO_SAFETY 22027
# - without Airspeed Sensor
param set-default CBRK_AIRSPD_CHK 1612128

```

If there isn't a more permanent solution available, I am willing to add the 'none_plane' and its definitions myself and create a pull request for it.

sfuhrer, 2023-12-15:
@Jaeyoung-Lim can you have a look here?

DiegoHerrera1890, 2025-02-18:
@alireza787b Thank you for sharing this. I am facing the same error. What version of PX4 did you use?

sfuhrer, 2025-02-18:
Here you have an example PR for adding a new SITL target: https://github.com/PX4/PX4-Autopilot/pull/23943/files

closed 2025-02-19 by mrpollo