PX4/PX4-Autopilot #22355: [Bug] RTL will sometimes not enter its land phase when  weathervane (`WV_EN`) is enabled

TedObrien, 2023-11-13:
### Describe the bug

In the event RTL is called, if weather-vaning causes the drones heading to change by more than `MIS_YAW_ERROR` during  the `RTL_STATE_CLIMB`,`RTL_STATE_DESCEND` and `RTL_STATE_RETURN` phases. The drone will remain at its return altitude indefinitely and not attempt to land.

### To Reproduce

To replicate in `sitl`:
- Start the windy gazebo environment: `make px4_sitl gazebo_iris__windy`
- Enable parameter  `WV_EN`
- Takeoff and call RTL while the drone is still trying to minimise its roll component 

### Expected behavior

Observe drone climb/decend to `RTL_RETURN_ALT` and remain there indefinitely

Console output will look something like this:

```
INFO  [navigator] RTL HOME activated	
INFO  [navigator] RTL: landing at home position.	
INFO  [navigator] RTL: climb to 492 m (4 m above destination)	
INFO  [navigator] RTL: return at 492 m (4 m above destination)

### Screenshot / Media

![Screenshot from 2023-11-13 15-53-57](https://github.com/PX4/PX4-Autopilot/assets/86834177/68b14690-e00d-4451-8753-03adbcee5cdf)

### Flight Log

https://review.px4.io/plot_app?log=5689aa08-0a97-4fb6-a140-5c92e81ad9e6

### Software Version

`v1.14`

### Flight controller

`sitl`, also tested on holybro v6x

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

The problem appears to be due to the yaw setpoint being based on `RTL_HDG_MODE`

https://github.com/PX4/PX4-Autopilot/blob/6cdf09644ea195de5cb0bb829e35ffeedbdbc86d/src/modules/navigator/rtl.cpp#L394-L413

If `WV_EN` is enabled then the heading may not be within the acceptance radius of the setpoint, hence the following line will not be satisfied, the `_rtl_state` will not advance and the drone will not start to descend.

https://github.com/PX4/PX4-Autopilot/blob/6cdf09644ea195de5cb0bb829e35ffeedbdbc86d/src/modules/navigator/rtl.cpp#L301

 I think a conditional to ignore the yaw setpoint in RTL mode if `WV_EN` is enabled will prevent this from happening. As a short term solution, setting `MIS_YAW_ERROR=180` appears to prevent this

danielhonies, 2023-11-16:
How does this work for normal mission items? If they have a yaw set point enabled and weathervane is on, will the vehicle not continue the mission. I think we should definitely get in this condition. Do you want to work out a PR?

TedObrien, 2023-11-16:
Just tried a mission with `WV_EN` enabled. 
- When the yaw is specified for a waypoint, the vehicle will not continue to the next one if weather-vaning its heading is outside of `MIS_YAW_ERR`. Also, if the hold time for a setpoint is 0, the vehicle oscillates around the setpoint. (https://review.px4.io/plot_app?log=3410fefb-bb99-418b-a781-f08e3622bcdb)
- Not specifying a yaw for waypoints means they can be completed.
- Setting `MIS_YAW_ERR` to 180 degrees allows the mission to complete when yaw is specified.

Happy to try and do a PR but may need a little guidance. Would overriding `MIS_YAW_ERR` if  `WV_EN` is enabled be sufficient? Or would ignoring yaw mission setpoints be preferred?

danielhonies, 2023-11-17:
I think it would be better to ignore yaw mission set points. Overriding parameters may lead to confusion.

TedObrien, 2023-11-20:
Created a PR #22413 , it would be great if you could take a look when you have a chance.

Thanks

TedObrien, 2023-12-21:
Fixed in #22532

closed 2023-12-21 by TedObrien