PX4/PX4-Autopilot #22356: [Bug]

AlexKlimaj, 2023-11-13:
### Describe the bug

I am getting a constant error tune playing without a PX4IO carrier board. 

Caused by https://github.com/PX4/PX4-Autopilot/pull/22243

```ERROR [px4io] PX4IO Not Supported
ERROR [px4io] PX4IO Not Supported
INFO  [tune_control] Stopping playback...
ERROR [px4io] PX4IO Not Supported
PX4IO start failed
```

### To Reproduce

Upload current main to a flight controller that doesn't include a PX4IO MCU.

### Expected behavior

No error tune on boot without a PX4IO.

### Flight Log

No Flight Log

### Software Version

Current main

### Flight controller

ARKV6X

### Vehicle type

None

julianoes, 2023-11-13:
Which target is this? I didn't know you could have the ARKv6X without IO.

AlexKlimaj, 2023-11-13:
> Which target is this? I didn't know you could have the ARKv6X without IO.
> 
> The check is based on whether the IO update file is included:
> 
> https://github.com/PX4/PX4-Autopilot/blob/00568985c0dd750d00eaaa53f05733e8f79714f0/ROMFS/px4fmu_common/init.d/rcS#L278

The IO MCU is on the carrier board. You can use any of the PAB form factor flight controllers on carrier boards with or without the IO. The ARK PAB Carrier and ARK Jetson PAB Carrier don't include the IO MCU. But you can use the ARKV6X on any of the Holybro or CUAV carriers with an IO MCU.

julianoes, 2023-11-14:
Right, so essentially the IO seems to be an option. So that means we need to make it "optional" which also means we can't raise an alarm if IO is not discovered when it should be there. We can't have both, bit messy.

AlexKlimaj, 2023-11-14:
We could extend the ver command to specially be able to query if the IO should be started.

closed 2023-11-20 by davids5