PX4/PX4-Autopilot #22357: Terrain Estimator not working as intended on main/RC2

ryanjAA, 2023-11-14:
### Describe the bug

This was actually tested on RC2 but they should be near identical to main.

This log is of a flight done last week where both the global position terrain estimate  and the estimator global position iwould toggle the entire flight. The issue was then terrain data would timeout and you couldn't auto land. This was fixed by altering the flight plan but that's not a real solution. https://review.px4.io/plot_app?log=eba74e3c-bc58-4cf6-84d2-de9034a0ecbe

If you look at this log, the global position terrain estimate just goes true false the entire flight. The estimator global position is a little more stable on it but by no means mirrored. https://review.px4.io/plot_app?log=4a862996-18a1-4d92-8022-0d219fa2cf95

Going back slightly in versions (just PR, still on 1.14) and both terrain messages follow identically without waiver (and if need be I can provide logs of the same location, almost same flight plan on 1.14 main).https://review.px4.io/plot_app?log=eba74e3c-bc58-4cf6-84d2-de9034a0ecbe

[image]

So what's going on here? This is RC2 with the RSSI from Ark added in and lidar sf30 driver and the i2c software reset. And the i2c reset pr has been added and checked extensively without issue. Logs available if needed.

Thanks

### To Reproduce

To reproduce you need a terrain estmate 

### Expected behavior

Terrain estimate to stay valid the entire time...

### Flight Log

https://review.px4.io/plot_app?log=eba74e3c-bc58-4cf6-84d2-de9034a0ecbe

### Software Version

1.14 RC2

### Flight controller

Pixhawk 6c

### Vehicle type

Fixed Wing

sfuhrer, 2023-11-14:
Thanks for the report! @bresch fyi

ryanjAA, 2023-11-14:
Further tested this just now.

Three logs, all in SITL. Same flight plan.

1 - [Flight Review](https://review.px4.io/plot_app?log=084931c7-4a69-4c28-8721-0881ac994677) Firmware with RSSI added in, SF30 driver and i2c reset. Based on 1.14 RC2 so only a few days prior to release but tested for good measure. Fails with terrain estimate valid/invalid (what was used in above flights not in SITL as well.

2 - [Flight Review](https://review.px4.io/plot_app?log=f5d7e895-f21c-4cf8-9957-2b0caf55ae9c) 1.14 Release (untouched). Same behavior as above, terrain estimate valid/invalid.

3 - [Flight Review](https://review.px4.io/plot_app?log=76f8ab88-738c-4a14-a9f7-dd87dd617366) 1.14 Main (from 2 days ago) - Correct behavior.

bresch, 2023-11-20:
@ryanjAA This might be the missing fix on 1.14: https://github.com/PX4/PX4-Autopilot/pull/22117

ryanjAA, 2023-11-21:
Thanks @bresch it does work in main but trying to test this against 1.14 release causes:

```
 fatal error: reference to non-static member function must be called; did you mean to call it with no arguments?
                                                                || (sq(_state.vel(0)) + sq(_state.vel(1)) > fmaxf(P.trace<2>(State::vel.idx), 0.1f));
```

which is one of the additions https://github.com/PX4/PX4-Autopilot/blob/c7672fc68c334121f8a41141a137b33a68da6bd4/src/modules/ekf2/EKF/range_height_control.cpp#L67
 so what's the best way to get around that or test against release?

Also this is changed (below). Since this should be be back ported or added to release version bump, just want to make sure we can do some testing, even if just sitl initially. 

https://github.com/PX4/PX4-Autopilot/blob/b8c541dd7277ed735139d7d1bfb829d61fbe29fb/src/modules/ekf2/EKF/position_bias_estimator.hpp#L38

vs 

https://github.com/PX4/PX4-Autopilot/blob/c7672fc68c334121f8a41141a137b33a68da6bd4/src/modules/ekf2/EKF/position_bias_estimator.hpp#L38-L39

bresch, 2023-11-21:
@ryanjAA I made a patch here: https://github.com/PX4/PX4-Autopilot/pull/22423

ryanjAA, 2023-11-21:
Tested via SITL quite a bit- works as intended. Will close and reopen if issues after testing live but don't anticipate any. 

Thanks @bresch :)

closed 2023-11-21 by ryanjAA