PX4/PX4-Autopilot #22359: [Bug] When fusing vision pose data, typhoon h480 experiences severe yaw setpoint change out of OFFBOARD modeclosed

Shawn-Ricardo, 2023-11-14:
### Describe the bug

sdfg

### To Reproduce

asdf

### Expected behavior

asdf

### Screenshot / Media

asdf

### Flight Log

asdf

### Software Version

asdf

### Flight controller

sadf

### Vehicle type

None

### How are the different components wired up (including port information)

asdf

### Additional context

sdf

Shawn-Ricardo, 2023-11-14:
Had asked in PX4 [discussions ](https://discuss.px4.io/t/rotation-in-yaw-when-entering-auto-land/33729)a few months ago, with no resolution from the community

Shawn-Ricardo, 2023-11-16:
Hi @beniaminopozzan , do you have any guidance on the issue above or is there someone else to possibly ask for support?

beniaminopozzan, 2023-11-18:
Hi @Shawn-Ricardo , I don't sorry. This is an "old" issue (#18781). My guess is that it is related to something in the EKF but I never have time to investigate it further.