PX4/PX4-Autopilot #22360: Rover Parameters Missing on Pixhawk 6X [Bug]

mcuomo1, 2023-11-14:
### Describe the bug

When running PX4 rover on the pixhawk 6X certain rover parameters do not exist such as gnd_cruise which sets the speed of the rover.  The rover also does not move when placed into mission mode.  The same firmware/ configuration works on the pixhawk 5X and the rover parameters exist and the vehicle navigates in mission mode.

### To Reproduce

Flash latest rover firmware to pixhawk 6x and search gnd_cruise in parameter list.  The gnd parameters are missing and the vehicle will not respond when switched into mission mode.

### Expected behavior

The rover parameters should appear when generic rover frame type is selected on the pixhawk 6x.

### Flight Log

NA

### Software Version

v1.14

### Flight controller

Pixhawk 6x

### Vehicle type

Rover

eogks8921, 2023-11-22:
Rover module is missing when building version 1.14.0 rc-1 using pixhawk6X. 1.14.0 stable version is also the same. Adding it will make it work. #22130

AlbertoJimenezUC3M, 2024-02-28:
> Rover module is missing when building version 1.14.0 rc-1 using pixhawk6X. 1.14.0 stable version is also the same. Adding it will make it work. #22130

Do we have to wait for a new release? Or have you been able to add it?

AlbertoJimenezUC3M, 2024-02-28:
Hello, has anyone been able to solve it? I can´t use a rover with 1.14 :(