PX4/PX4-Autopilot #22362: [Bug] Bug report on pre-checking geofence and mission uploads

czbxzm, 2023-11-14:
### Describe the bug

**PX4 adds safety mechanisms to ensure that the mission does not break the geofence.**
When a user uploads a mission, PX4 checks to see if the mission conflicts with geofence, and if it does it denies the mission from being uploaded to the aircraft.
[image]

But when this order is reversed and mission is uploaded before geofence, this check fails.
However, there are two logical flaws in the mechanism.
 **1.** When swapping the order, uploading the mission waypoint routes first and then plotting the no-fly zones does not trigger any checks, even if there is an overlap between the two;
 **2.** Even if we followed the correct order, the pre-check will only check if the waypoint is in the no-fly zone, and will not take into account the case where the route flies over the no-fly zone, like described in #22255.

### To Reproduce

### flaw1
1.upload the mission first;
2.draw a geo-fence and make the no-fly zone include the mission's route ;
3.uploaded the **geofenc** and found that it doesn't trigger a check warning as expected;
### flaw2
1.upload the geofence first;
2.Plot mission routes, but keep waypoints out of the no-fly zone and routes through the no-fly zone;
[image]

3.uploaded the **mission** and found that it doesn't trigger a check warning as expected;
4.Drones are allowed to take off as usual and fly through the no-fly zone on missions.

### Expected behavior

The pre-check mechanism should work properly to deny mission routes that violate the no-fly zone policy from being uploaded.

### Flight Log

NONE

### Software Version

<=1.14.0

### Flight controller

QGC

### Vehicle type

None

sfuhrer, 2023-12-15:
>1. When swapping the order, uploading the mission waypoint routes first and then plotting the no-fly zones does not trigger any checks, even if there is an overlap between the two;

Would you expect it to reject the geofence from getting uploaded in that case? Or would you invalidate the mission? And why would you need to update the geofence when already in-air?

czbxzm, 2023-12-16:
> Would you expect it to reject the geofence from getting uploaded in that case? Or would you invalidate the mission? 

I expect the mission feasibility check function to work correctly after updating both geofence and mission, since they are both updated after clicking the same upload button.There are many reasons to update geofence.

>And why would you need to update the geofence when already in-air?

When users are piloting a drone in-air and identify a hazardous area, such as a forest or airport, they are more likely to add a geofence while the drone is airborne, rather than landing and uploading the geofence. This behavior is also permitted by PX4.

sfuhrer, 2023-12-22:
>, since they are both updated after clicking the same upload button

I think this by itself is also an issue, for me geofence and mission are two completely separate things.