PX4/PX4-Autopilot #22373: [Bug] Unverified Home Point in No-Fly Zone

BladeY1, 2023-11-15:
### Describe the bug

In SITL simulation, we observed that the Home point can be arbitrarily set by the operator. Since Return to Launch (RTL) navigates the drone back to the Home point, inadvertent placement of the Home point within a no-fly zone can result in the drone breaching no-fly zone checks or entering the restricted area when the operator manually selects RTL. We validated this behavior using a drone equipped with PX4 flight control. It indeed demonstrated the ability to enter a no-fly zone through the aforementioned actions.

We believe that operators typically intend to avoid their drones entering no-fly zones to prevent potential damage or safety incidents. No-fly zones, often manually designated by operators, are crucial in areas like airports, amusement parks with high-altitude ride systems, or locations with dense obstacles such as forests, lakes, and clusters of buildings.

### To Reproduce

1. Start the drone and arbitrarily draw a no-fly zone.
2. Initiate drone takeoff (you can set up a mission). In the QGroundControl (QGC) flight interface, set the Home point within the designated no-fly zone.
3. Have the drone approach the edge of the no-fly zone (GFActionParam set to RTL mode) or manually trigger RTL mode as operated by the operator.

### Expected behavior

The drone disregards no-fly zone checks and enters the restricted area.

### Screenshot / Media

First, we draw a no-fly zone
[image]
Next, we command the aircraft to take off by uploading a mission. Slide to take off in QGroundControl (QGC)
[image]
Set the Home point within the no-fly zone.
[image]
Select 'Return' in QGroundControl (QGC).
[image]
Even with the triggered geofence check and the RTL mode activated upon fence breach, the drone ultimately enters the no-fly zone
[image]

### Flight Log

For unknown reasons, clicking the 'Upload' button on the log submission webpage yields no response. Additionally, there is no way to paste the logs or compressed files here.

### Software Version

HW arch: PX4_SITL
PX4 git-hash: 83841d1ad183e1b874fad1d07e5cd6a448d0c90e
PX4 version: 1.14.0 0 (17694720)
PX4 git-branch: main
OS: Linux
OS version: Release 5.15.0 (84869375)
Build datetime: Nov 14 2023 17:45:03
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.4.0
PX4GUID: 1006554d49534954414c44494e4f30303030
UNKNOWN MCU

### Flight controller

Open-Source DIY Drone

### Vehicle type

Multicopter

BladeY1, 2023-11-15:
We conducted actual tests on the DJI drone at the edge of a no-fly zone. When attempting to set the Home point within the no-fly zone using the Ground Control Station (GCS), DJI rejected the request and provided a prompt. DJI performs checks on the Home point setting, and we believe this check is necessary.

Drone-Lab, 2023-11-22:
@sfuhrer Hello,Did you fix this issue in #22394 new feature 4?
"do not allow to upload GF if it doesn't contain Home (to not break RTL)"

sfuhrer, 2023-11-28:
> @sfuhrer Hello,Did you fix this issue in #22394 new feature 4? "do not allow to upload GF if it doesn't contain Home (to not break RTL)"

Only partially. I didn't add logic to prevent a user to update the Home position when it would breach the geofence, but only the other way around, to reject a GF that doesn't contain Home.

What you ask for is now trivial as the part of the controller who does the Home position updating (Commander, see https://github.com/PX4/PX4-Autopilot/blob/8a68a66203d6975c2716a79e07ba56e37c06eb68/src/modules/commander/Commander.cpp#L951) has currently no idea about the geofence (which is in Navigator). I don't have a good idea right now how to do it, but if you have let us know!
PS do you even need the ability to manual update the Home position?

Drone-Lab, 2023-12-01:
> > @sfuhrer您好，您在#22394新功能4中修复了这个问题吗？“如果 GF 不包含 Home，则不允许上传 GF（以免破坏 RTL）”
> 
> 只是部分。我没有添加逻辑来阻止用户在违反地理围栏时更新 Home 位置，而是相反，拒绝不包含 Home 的 GF。
> 
> 您所要求的现在是微不足道的，因为控制器负责更新主位置（指挥官，请参阅
> 
> https://github.com/PX4/PX4-Autopilot/blob/8a68a66203d6975c2716a79e07ba56e37c06eb68/src/modules/commander/Commander.cpp#L951
> 
> ）目前不知道地理围栏（位于导航器中）。我现在不知道该怎么做，但如果您已经告诉我们！
> PS，您是否需要手动更新主页位置的功能？

> Only partially. I didn't add logic to prevent a user to update the Home position when it would breach the geofence, but only the other way around, to reject a GF that doesn't contain Home.
>
> What you [...]

Hi, @sfuhrer ,In the submitted #22469, the second point addresses the issue of geofence breach checks for the home point not being possible in commander.cpp due to geofence not being available in the navigator. The proposed solution is: **when the drone switches to RTL mode, check whether the home point triggers a geofence breach.** We would like to know if you find this reasonable. Currently, our commit successfully performs the check, but we haven't implemented event handling or actions after the check.
Regarding the detection of the home point within a no-fly zone, we believe that in such a scenario, the drone should switch to the Hold mode and wait for operator commands. Do you think this is reasonable? We look forward to receiving your advice.

BladeY1, 2023-12-01:
> Hi, @sfuhrer ,In the submitted #22469, the second point addresses the issue of geofence breach checks for the home point not being possible in commander.cpp due to geofence not being available in [...]

The approach seems reasonable. It's somewhat similar to the strategy Autel uses for no-fly zones in our tests (allowing setting the home point in a no-fly zone but preventing RTL return). Although DJI's strategy of disallowing setting the home point in a no-fly zone might be more logical, implementing it in the commander is challenging. I attempted to add a no-fly zone check in homeposition.cpp's setmanual, but I couldn't obtain the navigator pointer. However, considering rally points (e.g., a valid rally point being closer to the drone and the home point being invalid), checking the home point should be performed after determining the need to use the home point and then verifying if it triggers a geofence breach.

sfuhrer, 2023-12-15:
I challenge the need to update Home after already in air. If the position to make the RTL towards has to be changed in-air the user should set RallyPoints - for which we btw also need to add a geofence check (prevent the uploading of a RallyPoint that breaches the GF).

Drone-Lab, 2023-12-21:
> I challenge the need to update Home after already in air. If the position to make the RTL towards has to be changed in-air the user should set RallyPoints - for which we btw also need to add a geofence check (prevent the uploading of a RallyPoint that breaches the GF).
> 
> > The proposed solution is: when the drone switches to RTL mode, check whether the home point triggers a geofence breach.
> 
> And what's your proposed action then? Go to Hold mode instead of RTL? That's quite dangerous no? RTL is the mode that has to bring the drone automatically back if communications are lost for example, in which case Hold would result in a crash after the battery runs out.

We discussed the following and have come up with a solution.

Firstly, regarding your mention of "the need to update Home after already in the air," we believe that, for operators, modifying the home point in the flight interface may be more straightforward than setting a rally point in the mission interface and then uploading it. Therefore, many operators may prefer to choose to modify the home point for the return.

Currently, the solution we are considering is to save the original home point of the drone and set it as the original point. When it is detected that the home point is within the geofence, the system will hold for 5 or 10 seconds and prompt the operator to take action. If the operator does not take control, the drone will return to the original home point. Of course, a potential issue with this solution is that the drone may not be able to return to the original point due to low battery. However, if the operator fails to take control during the hold and return process for such a long time, leading to insufficient battery and a forced descent that could result in a potential crash, we consider this to be an issue of operator negligence. We have already provided the maximum safety guarantee.

Of course, the best solution would be to directly prohibit setting the home point within a no-fly zone.