PX4/PX4-Autopilot #22374: [Bug] Seemingly incorrect setpoint in followme mode causing a fly-off

Fraisa, 2023-11-15:
### Describe the bug

Was testing follow-me on our drone initially it was all good, but then suddenly the drone took off (towards a power-line some way off) and I had to abort and enter position mode. 

Two issues was noted from our perspective. 

- In follow me drone received a set-point that was about 200m away from our position, this could have been from the Herelink GPS or an internal issue - I do not know, but the error was large and thus I do not think it was a GPS error.
- Once I entered position mode, it continued towards the setpoint it previously had from Follow Me mode. I think it should once you enter position mode reset the set-point and  stop where it is. This could be an error on my part.

### To Reproduce

(Not yet reproduced it but here is what I did)

1. Remote switched on.
2. Drone switched on.
3. Took off in position mode .
4. Swapped to Follow Me mode and walked around to see how it reacts. Noted too aggressive settings for a drone our size.
6. Noted the drone was taking off from us towards the power lines, and swapped to position mode.
7. Noted drone was NOT stoping and manually took control over it to stop it. 

### Expected behavior

Firstly, the set point in follow me mode should not be so far from us even if you have a very bad GPS in the ground statios. 

Secondly, when entering position mode from follow me mode, any set point should be clearerd and current position should be held. (This could be an incorrect assumption on my part.) 

### Flight Log

https://logs.px4.io/plot_app?log=00d1f070-bc99-4a03-8690-1ac7225a28ee

### Software Version

Used Px4 1.14 stable release. 

### Flight controller

CUAV V5+

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

I don't think this is relevant.

### Additional context

I won't be able to reproduce this for a while as the normal test field is too small to test safely but hope to do so in time.

junwoo091400, 2023-11-16:
Seems like Y coordinate of position setpoint went crazy low, I will check later why that may have happened

Fraisa, 2023-11-20:
Thank you, any update would be greatly appreciated!

suezere, 2023-12-20:
Hi I’m trying to fly by building V1.14 on Cuav v5+. But here comes the problem. After building the beta version, the FC status color changes to red. I would like to ask if you have encountered this situation and how to solve it. Thanks.

Fraisa, 2024-01-17:
> Hi I’m trying to fly by building V1.14 on Cuav v5+. But here comes the problem. After building the beta version, the FC status color changes to red. I would like to ask if you have encountered this situation and how to solve it. Thanks.

I would need more information to understand what you mean exactly, but I used 1.12 previously and needed to setup the actuators when swapping to 1.14 before the FC stopped being red, that's maybe not your issue but the only "FC went red" issue I've noted on.