PX4/PX4-Autopilot #22382: [uxrce_dds_client] RTT too high for timesync

dakejahl, 2023-11-16:
### Describe the bug

Just updated to main and now I am seeing this message every 3.5 seconds on the dot.
```
WARN  [timesync] RTT too high for timesync: 24 ms
```

I'm running the bridge on serial at 921600. 

### To Reproduce

Flash main

### Expected behavior

No periodic warnings

### Screenshot / Media

WARN  [timesync] RTT too high for timesync: 18 ms
WARN  [timesync] RTT too high for timesync: 18 ms
WARN  [timesync] RTT too high for timesync: 24 ms
WARN  [timesync] RTT too high for timesync: 24 ms
WARN  [timesync] RTT too high for timesync: 23 ms
WARN  [timesync] RTT too high for timesync: 18 ms
WARN  [timesync] RTT too high for timesync: 19 ms
WARN  [timesync] RTT too high for timesync: 18 ms
WARN  [timesync] RTT too high for timesync: 19 ms
WARN  [timesync] RTT too high for timesync: 19 ms
WARN  [timesync] RTT too high for timesync: 18 ms
WARN  [timesync] RTT too high for timesync: 19 ms
WARN  [timesync] RTT too high for timesync: 22 ms
WARN  [timesync] RTT too high for timesync: 18 ms
WARN  [timesync] RTT too high for timesync: 22 ms
WARN  [timesync] RTT too high for timesync: 28 ms
WARN  [timesync] RTT too high for timesync: 18 ms
WARN  [timesync] RTT too high for timesync: 14 ms
WARN  [timesync] RTT too high for timesync: 23 ms

### Flight Log

no log

### Software Version

main

### Flight controller

ark_fmu-v6x_default

### Vehicle type

None

dakejahl, 2023-11-20:
Might be related to https://github.com/PX4/PX4-Autopilot/issues/22399

tstastny, 2023-11-20:
@bkueng

closed 2024-04-14 by beniaminopozzan